Koenkk/zigbee2mqtt #28504: Bosch BTH-RM Thermostat not exposing entities in Home Assistant despite successful pairing

MKellaz, 2025-09-05:
### What happened?

Hi there,
I’ve successfully paired a Bosch BTH-RM room thermostat with Zigbee2MQTT. The device appears correctly in the Zigbee2MQTT frontend, and I can see all expected values such as local_temperature, occupied_heating_setpoint, system_mode, etc.
However, in Home Assistant, no entities are created for this device — not even under the MQTT integration. Other Zigbee2MQTT devices work fine and show up as expected.

It seems the device is sending data, but Zigbee2MQTT does not expose it in a format that Home Assistant can discover automatically. Could this be due to a newer firmware version on the device that isn’t fully supported yet?

The installed firmware on the device is newer then the available version from Zigbee2MQTT.

[image]

### What did you expect to happen?

I would expect that the devices and entites show up in Home Assistant.

### How to reproduce it (minimal and precise)

Connect a Bosch BTH-RM with firmware 0.2.09-6.a.90 to Zigbee2MQTT.

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

7.4.4

### Adapter

EmberZNet

### Setup

Add-on on Home Assistant OS in a ProxMox Virtual Machine.

MKellaz, 2025-09-07:
I think i found something:

The device is successfully paired and shows up in Home Assistant, but only as a climate entity. It does not appear as a full device under the MQTT integration, which I believe is not entirely correct. Other Zigbee2MQTT devices (like the Bosch radiator thermostat) show up as proper devices with grouped entities, so I expected similar behavior here.

Functionally, the BTH-RM shares many sensors and attributes with the Bosch radiator thermostat - such as local_temperature, occupied_heating_setpoint, system_mode, humidity, boost_heating, and window_detection. However, it doesn’t control a heating actuator directly, which makes sense.

I’m planning to write a custom converter to improve the integration and expose the missing attributes properly.  I’m an application developer, mostly familiar with C# and Windows applications — JavaScript/TypeScript is far outside my comfort zone 😆 

Thanks in advance for any guidance. Once I get it running properly, I’d be happy to take a shot at a PR to improve the converter 😉

LouisHaftmann, 2025-09-27:
I have the same issue. Paired successfully and all the metrics are showing up in zigbee2mqtt but no HA device.

MKellaz, 2025-09-27:
> Could you provide the debug log when starting z2m and the data/database.dv entry of this device?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Hi @Koenkk , i will attach the logfile but where do i find the data/database.dv?

Could the firmware be part of the issue? Since last week (?), Zigbee2MQTT has been showing my installed firmware version as the available one. Before that, my installed version was actually newer than the one listed in Z2M (see the screenshot in my first post). Also, the “firmware ID” looks suspicious to me 🤔 
@DerDreschner, what version do you have installed and do you see a valid "firmware ID"?

[image]

The Logfile:
[45df7312_zigbee2mqtt_2025-09-27T20-44-18.559Z.log](https://github.com/user-attachments/files/22576747/45df7312_zigbee2mqtt_2025-09-27T20-44-18.559Z.log)

Koenkk, 2025-09-28:
It looks to be discovered, so I'm not sure if this is an issue with your HA setup (e.g. `Received MQTT message on 'homeassistant/climate/0x001e5e0902ad1a29/climate/config' with data ` contains this discovery). You can also try to remove the device, restart z2m and re-pair it.

DerDreschner, 2025-09-29:
@Koenkk: Hmm, I can confirm that there is something wrong with the current stable version of Z2M when using it together with the BTH-RM. In my case, it doesn't want to show up in Home Assistant at all. I wasn't able to find the reason for that. But I didn't experienced that issue with Z2M Edge, so... Lets wait and see if it's solved for @MKellaz and @BastiKur as well?

zeratul021, 2025-09-29:
FYI I am using both thermostats (battery and 230V version) for cca 3 months with up-to-date addon version of z2m in HA and I see the same entities.

> Hey there,
> 
> I didn't experienced the issue you have. The current code exposes a lot more functions then just the thermostat. For example, this is how the BTH-RM is currently working in my HA instance (without any changes from my side yet):
> 
> <img alt="Image" width="1043" height="1291" src="https://private-user-images.githubusercontent.com/34408442/494470102-9765d172-899a-41f9-994d-bd50ae224476.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxODE1NjIsIm5iZiI6MTc1OTE4MTI2MiwicGF0aCI6Ii8zNDQwODQ0Mi80OTQ0NzAxMDItOTc2NWQxNzItODk5YS00MWY5LTk5NGQtYmQ1MGFlMjI0NDc2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDIxMjc0MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTYyOWJjODYyM2E1ZTY3NWEzZTAwMmRjYTYwMTE4N2VmZDg1ZjY5ZTk1ZWJiNzdkOWU5NjA0ODUyODIzY2YzOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.cIOKMAIuHA-sj82ppPtfTtAd21d0Rg0iWHzli_IxYgE">
> But could it be that you had some issues with pairing?
> 
> Greetings!

BastiKur, 2025-10-02:
I just updated Zigbee2MQTT to version 2.6.2-1 and everything works. Since I force-deleted one of the room thermostats, I had to pair it using the install code. But the other two were already listed in Z2M. For both of them I just clicked on "Device Interview" and "Reconfiguration" and now all three room thermostats show up in MQTT along with 13 entities.
Happy times.