Koenkk/zigbee2mqtt #28507: my prevoisly working tuya door windows contact stop working. Not the devices appears as isense  sensor (TUYA TS 0203)

xEn64, 2025-09-05:
### What happened?

Update-> Door Window Conact stop working

### What did you expect to happen?

it should work

### How to reproduce it (minimal and precise)

TUYA TS 0203 install on 2.6.0.1, then patch to 2.6.1.1 -> devices stop working and gets as isense in the system

### Zigbee2MQTT version

2.6.1.1

### Adapter firmware version

2025

### Adapter

SLZ06

### Setup

generic x86

### Device `database.db` entry

{"id":80,"type":"EndDevice","ieeeAddr":"0xa4c138edbf201d2d","nwkAddr":37317,"manufId":4742,"manufName":"Zbeacon","powerSource":"Battery","modelId":"TS0203","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"ssIasZone":{"attributes":{"61441":2,"iasCieAddr":"0xf4b3b1fffee1e530","zoneState":1,"currentZoneSensitivityLevel":1}},"genPollCtrl":{"attributes":{"checkinInterval":6960}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":28}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xf4b3b1fffee1e530","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0},{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":65000,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1371223903},"lastSeen":1757083778054,"checkinInterval":1740}

jopereira, 2025-09-05:
I can confirm this. After the update from 2.6.0 to 2.6.1 the door/window sensor is now detected as a temperature/humidity sensor.

Koenkk, 2025-09-05:
Should already be fixed in the dev branch.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-09-05 by Koenkk