Koenkk/zigbee2mqtt #28521: [lumi.sensor_ht.agl001 / Aqara] Temp threshold modification not possible

aldoushx, 2025-09-07:
### What happened?

The modification of the temperature threshold is not kept if set under 0.5 °C. When the threshold is set under 0.5°C (0.2 or 0.3 or 0.4 °C), it returns  automatically to 0.5°C, with no error in the log, the payload is sent without error. It works with thresholds greater than 0.5, like 0.7 or 1°C.
When the report by period is set, the temperature is returned with a temperature step much smaller than 0.2 °C.

Same with temperature and humidity periods, which cannot be set back to the original walue of 30s after modification. The value is now limited to 10s, the max value of the bracket in the frontend.

### What did you expect to happen?

To modify the temperature threshold to 0.2°C.

### How to reproduce it (minimal and precise)

Reduce the temp_threshold to 0.2°C in the frontend or in HA, it switches back automatically to 0.5°C.

### Debug logs
[Zigbee2MQTT](http://192.168.1.36:8080/#/)

[07/09/2025 09:51:38] frontend:api: Sending {"topic":"0x54ef4410013b0547/set","payload":{"temp_threshold":0.2}}

[07/09/2025 09:51:43] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Thermostat chambre amis 2', payload '{"display_off":false,"external_humidity":null,"external_temperature":null,"high_humidity":99.99,"high_temperature":50,"humi_period":10,"humi_report_mode":"threshold","humi_threshold":5,"humidity":61.45,"identify":null,"linkquality":200,"low_humidity":0,"low_temperature":0,"period":0.5,"sampling":"standard","sensor":"external","temp_period":10,"temp_report_mode":"threshold","temp_threshold":0.2,"temperature":25.51,"update":{"installed_version":-1,"latest_version":-1,"state":null}}'

[07/09/2025 09:51:43] z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Thermostat chambre amis 2', payload '{"display_off":false,"external_humidity":null,"external_temperature":null,"high_humidity":99.99,"high_temperature":50,"humi_period":10,"humi_report_mode":"threshold","humi_threshold":5,"humidity":61.45,"identify":null,"linkquality":200,"low_humidity":0,"low_temperature":0,"period":0.5,"sampling":"standard","sensor":"external","temp_period":10,"temp_report_mode":"threshold","temp_threshold":0.5,"temperature":25.51,"update":{"installed_version":-1,"latest_version":-1,"state":null}}'

### Details
os: `#1 SMP PVE 5.15.143-1 (2024-02-08T18:12Z) - 5.15.143-1-pve - x64`
node: `v20.18.2`
zigbee2mqtt: `2.6.1` (`d44463a3`)
zigbee-herdsman: `6.0.4`
zigbee-herdsman-converters: `25.14.0`
adapter: `EmberZNet` `{"build":0,"ezsp":13,"major":7,"minor":4,"patch":5,"revision":"7.4.5 [GA]","special":0,"type":170}`
#### Device
software_build_id: `undefined`
date_code: `20250507`
endpoints:
```json
{"1":{"bindings":[{"cluster":"msTemperatureMeasurement","target":{"endpoint":1,"ieee_address":"0xe0798dfffeb78812","type":"endpoint"}},{"cluster":"msRelativeHumidity","target":{"endpoint":1,"ieee_address":"0xe0798dfffeb78812","type":"endpoint"}}],"clusters":{"input":["genMultistateInput","msRelativeHumidity","msTemperatureMeasurement","genPowerCfg","genIdentify","genBasic","manuSpecificLumi"],"output":["genOta"]},"configured_reportings":[{"attribute":"measuredValue","cluster":"msTemperatureMeasurement","maximum_report_interval":3600,"minimum_report_interval":10,"reportable_change":100},{"attribute":"measuredValue","cluster":"msRelativeHumidity","maximum_report_interval":3600,"minimum_report_interval":10,"reportable_change":100}],"name":"plus","scenes":[]},"2":{"bindings":[],"clusters":{"input":["genMultistateInput"],"output":[]},"configured_reportings":[],"name":"center","scenes":[]},"3":{"bindings":[],"clusters":{"input":["genMultistateInput"],"output":[]},"configured_reportings":[],"name":"minus","scenes":[]}}
```
##### Health
time: `Sun Sep 07 2025 09:06:35 GMT+0200 (heure d’été d’Europe centrale)`
process.uptime_sec: `84006`
```json
{"leave_count":0,"messages":622,"messages_per_sec":0.0154,"network_address_changes":0}
```

MoeRT09, 2025-11-03:
Hey, 
I am able to set the temp threshold to 0.2 °C in zigbee2mqtt v2.6.3.
However my device firmware date code is `20250922` so maybe it gut fixed in firmware (just switch from thread to zigbee Firmware today via aqara app.

But could it be possible that the maximum / minimum boundaries for the temp and humidity report periods are defined wrong?
Doesn't really make sense to limit the maximum report interval to 10 seconds while the temperature and humidity sampling period can be set to 600 s. The other way around would make more sense.

I need my thermostats remote temperature set at least once per 30 minutes or it will fall back to the internal sensor. With only threshold reporting enabled in the w100 I observed the thermostat falling back to internal sensor because no temperature change was reported by the w100 due to marginal temperature changes measured in the room. Now I needed to switch it to threshold and period reporting. However with the temperature being reported every 10 seconds I am worried about the battery life of the w100 and the thermostats.

hagih, 2025-11-03:
@MoeRT09 

> Hey, I am able to set the temp threshold to 0.2 °C in zigbee2mqtt v2.6.3

Are you sure the value is actually sticking? I can set it to 0.2°C as well but after a second or two it changes back to 0.5°C on it's own. I am also using `Zigbee2MQTT 2.6.3-1`. I don't see firmware `20250922` for CC2652P, perhaps you are using a different radio?

> But could it be possible that the maximum / minimum boundaries for the temp and humidity report periods are defined wrong? Doesn't really make sense to limit the maximum report interval to 10 [...]

I am absolutely convinced the value **limits are wrong**. In Zigbee2MQTT the Temp period has a minimum value of 1 second and a maximum value of 10 seconds! ([link](https://www.zigbee2mqtt.io/devices/TH-S04D.html#temp-period-numeric)). This is ludicrous. Imagine the battery usage if it sent a zigbee signal every 1 second. In the Aqara app, the _minimum_ temperature period is 10s (not the maximum) and the values increase from there ([screenshot](https://itad-forum-ovs-1300889962.aqara.com/original/2X/8/8cb645423bb62f820a04769b7e1f42946d40ea7a.jpeg), [context](https://forum.aqara.com/t/w100-explanation-of-the-temperature-sampling-setting/52930))

hagih, 2025-11-03:
I believe I found the original author of the config code. I added comments there and tagged him to draw attention (and updated with your finding of a 10m/600 second possible max):

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9260/files#diff-11e8450decb6d1cc6caa9e7b7c76b5d53c34b34d5e6e30f1b6b55122a20c5e54R4672-R4673

aldoushx, 2025-11-03:
I updated it with the aqara app on my smartphone : reset the device, set it up with the thread firmware to erase the ZigBee firmware with the bluetooth procedure, then go back to the ZigBee firmware : il will be with the last ZigBee firmware. 

With this update I can now set the temperature threshold to 0,2°C, and the device keeps it. I am not on the last Z2mqtt version.

hagih, 2025-11-03:
Thank you very much for the info.

I can confirm that I followed your suggestion just now and it worked! I am now able to modify the temperature threshold under 0.5°C, thus the subject of this post is indeed resolved by a firmware update on the device itself.

For the benefit of anyone else coming across this thread, these are the exact steps I took:

- Reset the network settings of the Aqara W100 temperature sensor by pressing the reset button 10 times
- Opened the Aqara Home app on my phone and ensured it had permissions (note: an Aqara hub is not needed, just the app)
- The app detected the W100 immediately, I clicked it to begin adding and selected _Thread_ as the protocol when prompted. It flashed a Thread firmware and then after updating gave an error about not having a Thread border router
- Went back to the main menu in the app, clicked the W100 again and selected _Zigbee_. It flashed the firmware to the latest Zigbee
- Deleted the old device from Zigbee2MQTT using the force option
- Clicked Permit join in Zigbee2MQTT
- Device was immediately detected and added. Renamed it to my preference/the original name
- It is now possible to set the temperature threshold to 0.2°C in HA without it changing back!

closed 2025-11-03 by aldoushx