Koenkk/zigbee2mqtt #28525: Bosch BMCT-RZ does not pair completely.

ZhuraEN, 2025-09-07:
### What happened?

In the process of connecting the relay, I get an error:

```
error 2025-09-07 15:38:12zh:controller: Interview failed for '0x0cae5ffffe37993b with error 'Error: Interview failed because can not get node descriptor ('0x0cae5ffffe37993b')'
error 2025-09-07 15:38:12z2m: Failed to interview '0x0cae5ffffe37993b', device has not successfully been paired
```

### What did you expect to happen?

I expect the device to connect, as it is in the list of supported devices.

### How to reproduce it (minimal and precise)

During the standard pairing process, even after the factory reset, the device does not connect.

### Zigbee2MQTT version

2.6.0

### Adapter firmware version

20240710

### Adapter

CC2652P

### Setup

Run on Wirenboard 8.5 controller.

badr-ctrl, 2025-09-08:
Bosch devices require pairing by code

badr-ctrl, 2025-09-09:
i think you have to use it twice,one to pair one,and again to see the commands

ZhuraEN, 2025-09-09:
This is quite strange, how do you think it should happen? I enter the code, the device starts searching, it tries to take the interview, but it doesn't happen, resulting in an error. At what point should I enter the code a second time? To clarify, I have other devices from this ecosystem in my system, and they have not encountered any issues - they are successfully added.

ZhuraEN, 2025-09-09:
It was my mistake - I didn't check the code I was entering, and there was an error. The device was added