Koenkk/zigbee2mqtt #28562: [C-ZB-LC20v2-RGBCCT / Candeo] ???

litecross91, 2025-09-09:
### What happened?

Gledopto GL-C-002P was in GL-C-006P mode (CCT).
After automatic OTA update doesnt work. zigbee2mqtt shows Model
[C-ZB-LC20v2-RGBCCT](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)
[Unsupported](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)

### What did you expect to happen?

### How to reproduce it (minimal and precise)

### Debug logs

### Details
os: `#1 SMP PREEMPT_DYNAMIC Wed Aug 13 08:49:33 UTC 2025 - 6.12.41-haos - x64`
node: `v22.16.0`
zigbee2mqtt: `2.6.1` (`unknown`)
zigbee-herdsman: `6.0.4`
zigbee-herdsman-converters: `25.14.0`
adapter: `ZStack3x0` `{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20250321,"transportrev":2}`
#### Device
software_build_id: `30151203`
date_code: `20250723`
endpoints:
```json
{"11":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","genGroups","genScenes","genOnOff","genLevelCtrl","lightingColorCtrl","touchlink"],"output":["genOta"]},"configured_reportings":[],"scenes":[]}}
```
##### Health
time: `Wed Sep 10 2025 00:46:26 GMT+0200 (közép-európai nyári idő)`
process.uptime_sec: `103803`
```json
{"leave_count":0,"messages":5048,"messages_per_sec":0.0486,"network_address_changes":0}
```

Noirbonn666, 2025-09-10:
Same here with a C-ZB-LC20v2-RGBCCT. Run the Update it reported as v2 should be still the same Device still work ...

skyfullofdust, 2025-09-10:
Apparently the firmware came from this commit: https://github.com/Koenkk/zigbee-OTA/commit/f1b86e5c6edb961230a75d91d918437ab18b3031

peterthepeter, 2025-09-10:
@Fierys 

Thanks man this worked for me as well. 👍

davux, 2025-09-10:
I have a Gledopto GL-C-201P in RGB+CCT mode, currently recognized as GL-C-008P as the [device page](https://www.zigbee2mqtt.io/devices/GL-C-008P.html) says.

Not sure if OTA'ing to LC20 RGBCCT would work.

Based on visual clues however, the 201P **does** look more like the LC20 than the 008P (despite being from a different brand), so there's that.

Candeo C-ZB-LC20-RGBCCT:
[image]

Gledopto GL-C-201P:
[image]

litecross91, 2025-09-10:
I would also be interested in this. It seems that due to a fault, everyone who has these types of Gledopto devices, incorrectly received the OTA update intended for Candeo.... :

RGB+CCT GL-C-008P [White]
RGBW GL-C-007P [Yellow]
RGB GL-C-003P [Blue]
CCT GL-C-006P [Green]
Dimmer GL-C-009P [Red]
GL-C-001P - Zigbee Pro 5 in 1 Smart LED Controller
GL-C-002P - Zigbee Pro 5 in 1 LED Controller Mini Ultra Thin
GL-C-011P - Zigbee Pro 5 in 1 Smart LED Controller DIN Rail
GL-C-201P - Zigbee Pro+ 5 in 1 Smart LED Controller
GL-C-202P - Zigbee Pro+ 3 in 1 Smart LED Controller
GL-C-301P - Zigbee Pro+ 5 in 1 Smart LED Controller Ultra-Mini

See here: https://github.com/Koenkk/zigbee-OTA/pull/864

It's a good question how those who installed it will put the Gledopto firmware back.

drax-uk, 2025-09-10:
I have done the update as well on my GL-C-002P, and I managed to get it working after repairing, but also stuck with model C-ZB-LC20v2-RGBCCT and Not supported. Hope there is a way to force the downgrade back to the old firmware.

atl285, 2025-09-10:
> Definitely not better for me. I've lost the ability to control color temperature, and to change modes. I really want to downgrade my GL-C-002P back to the Gledopto firmware.

Did you tried the instructions from the [candeo page](https://support.candeo.io/portal/en/kb/articles/czblc20-how-do-i-pair-my-zigbee-led-controller-to-home-assistant-zigbee2mqtt)? After firmware update I had to use the **opt** button to switch back to the DIM mode (red led) and after this I have removed the device from Zigbee2MQTT (via hitting the trashcan button in the frontend) and activated the pairing in Zigbee2MQTT. After some minutes it appears as DIMable device again.

Koenkk, 2025-09-10:
I've disabled the OTA for now and will take action to make sure this cannot happen again

> I’m experiencing the same issue with the GL-C-002P.
> 
> I attempted to send a downgrade payload as described here: [Koenkk/zigbee-OTA#827](https://github.com/Koenkk/zigbee-OTA/issues/827). However, since the device model has changed, it now reports that OTA updates are not supported.
> 
> Is there any way to force a downgrade command?

To make z2m detect the device correctly:
- Stop z2m
- In the `data/database.db` change the modelId for this device to `GL-C-002P` (note that 2 occurrences have to be changed)
- Start z2m.

boardlord1, 2025-09-10:
> I've disabled the OTA for now and will take action to make sure this cannot happen again
>
>
> To make z2m detect the device correctly:
>
>     * Stop z2m
>
>     * In the `data/database.db` change the [...]

The Gl-C-002P changes names depending on the mode we put it in. I'm using it in simple WW/CW mode, so in the database.db I changed the manufName to Gledopto and the modelId to Gl-C-006P in both places. manufId seems to be the same for both Candeo and Gledopto so left that in place. Upon starting up Z2M it came up OK. Thanks!

boardlord1, 2025-09-10:
> The Gl-C-002P changes names depending on the mode we put it in. I'm using it in simple WW/CW mode, so in the database.db I changed the manufName to Gledopto and the modelId to Gl-C-006P in both [...]

So this works with manual tweaking, but what happens if we need to reinstall Z2M, or need to reset the device that has the new FW? Will we have to do this manual tweak every time? Or is a downgrade possible?

litecross91, 2025-09-10:
Now upload the GL-C-002P_V20551203_20240125.ota file. I don't know yet if it will work.

Tutorial:
1. ust upload the ota file to the /config/zigbee2mqtt/ota folder. 
2. Create the my_index.json file in the /config/zigbee2mqtt/ folder.
my_index.json:
```
[
  {
    "fileVersion": 620834817,
    "url": "ota/GL-C-002P_V20551203_20240125.ota",
    "imageType": 5145,
    "manufacturerCode": 4687,
    "force": true
  }
]
```
3. Edit the /config/zigbee2mqtt/configuration.yaml file. Add these two lines:
```
 ota:
  zigbee_ota_override_index_location: my_index.json
```
4. Restart Z2M. The update will then be available on the OTA tab.

boardlord1, 2025-09-10:
> [@boardlord1](https://github.com/boardlord1) for e.g. `GL-C-002P` you could try to force the OTA on your device.
> 
>     * Grab the image from https://github.com/Koenkk/zigbee-OTA/tree/master/not-in-manifest-images/Gledopto (I see there are three for `GL-C-002P`, not sure what the diffs are)
> 
>     * Push the OTA to the device using the [local OTA index](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files) (make sure to put `force: true`)

Worked perfectly, all good now even after re-interviewing the controller! Thanks

Marco4223, 2025-09-15:
> > Just make sure your device is the 5-in-1 Pro Mini (now rebranded as “ultra thin”):
> 
> [@diegocjorge](https://github.com/diegocjorge), this is exactly my device. After creating the `my_index.json` file and include it in the `configuration.yaml`, the device shows an upgrade. I will try it within the next days. Thanks. 😄

If you have success with this, would you be so kind as to describe what you changed, how and where so that I can understand it as a zigbee2mqtt newbie? I have three of these controls and am slowly going mad here
Thanks

waf1tb, 2025-09-15:
Ok now it had worked I had my controller on blue (rgb) and had to change it to GL-C-003P now it had worked.
Thanks to everyone who helped me and made the project possible.
I come from Germany and my English is unfortunately not so good and sometimes has a few hurdles.
Translated with AI
😊

[image]

fabietx, 2025-09-20:
> firmware downgrade failed. I created directory ota in /opt/zigbee2mqtt/data, copy downloaded GL-C-008P_V14_OTAV4_20210114_100.ota to ota/, chowned to root. Created my_index.json in /opt/zigbee2mqtt/data with
> 
> ```
> [
>     {
>         "fileVersion": 620834817,
>         "url": "ota/GL-C-008P_V14_OTAV4_20210114_100.ota",
>         "imageType": 5145,
>         "manufacturerCode": 4687,
>         "force": true
>     }
> ]
> ```
> 
> Edit Settings OTA updates under OTA index override file name my_index.json. stopped zigbee2mqtt. Edit database.db and changed "modelId":"C-ZB-LC20v2-RGBCCT" to "modelId":"GL-C-008P" and manufName from CANDEO to GLEDOPTO. Restared zigbee2mqqtt. Devices are recognized as GL-C-008P again and under OTA, after checking for new firmware, update device firmware I got z2m: Update of 'Schrankbeleuchtung_LED_Wz' failed (No image currently available) :-( What went wrong?

I got the same error ...(No image currently available) when checking the new firmware, but in the log I saw that the 'my_index.json' file was not formatted correctly (DOS encoding). Cleaning it with the linux command 'dos2unix' did the trick.

francocm, 2025-09-20:
> firmware downgrade failed. I created directory ota in /opt/zigbee2mqtt/data, copy downloaded GL-C-008P_V14_OTAV4_20210114_100.ota to ota/, chowned to root. 

@rstein5 and @fabietx 

Any reason why you're using `GL-C-008P_V14_OTAV4_20210114_100.ota` (quite old) instead of `GL-C-002P_V20551203_20240125.ota` (newer)?

Assuming I understood the way all this works (and so seems to have been the case in my experience here):

The `GL-C-008P` refers to `GL-C-002P` running in `RGB+CCT` mode, making the `GL-C-002P` identify itself as `GL-C-008P`. From what I understand the firmware should be the same regardless of mode it's running as, which leads me to think the one you picked is way too old potentially.

I've used `GL-C-002P_V20551203_20240125.ota` on three devices of mine, all `GL-C-002P`, but 2 identified as `GL-C-008P` (`RGB+CCT`) and 1 as `GL-C-006P` (`CCT`) in my case.

[image]

rstein5, 2025-09-20:
> Any reason why you're using `GL-C-008P_V14_OTAV4_20210114_100.ota` (quite old) instead of `GL-C-002P_V20551203_20240125.ota` (newer)?
>
> Assuming I understood the way all this works (and so seems to [...]
Thanks for the hint. The update is now running on the first device.
I guess I relied too much on the designation GL-C-008P and downloaded the wrong file.

fabietx, 2025-09-21:
@francocm I've used the last version proposed "GL-C-002P_V20551203_20240125.ota", the upgrade complete succesfully, so in a early phase I was happy to have fixed my devices but.... today I recognized that the physical wall switches don't work anymore :( In fact my controllers are GL-C-201P which have the switch contacts. Maybe the GL-C-002P firmware is not suitable for these.
So now I' m stucked again, if there is no public version of the original GLEDOPTO firmware for this kind of controllers I should install again the Candeo version that even it appears to not be supported in the GUI it worked. Is it possible?
I can use the same procedure? My only doubt is how to compile the my_index.json file with the correct "fileversion" and "imagetype".

Thanks in advance for any hints

litecross91, 2025-09-29:
[GL-C-002P_V20951203_20241017_32013001.zip](https://github.com/user-attachments/files/22593516/GL-C-002P_V20951203_20241017_32013001.zip)
I got this, but it doesn't work for me due to a timeout error. This is the latest version. Try it and see if it works for you. For GL-C-002P version.

Fierys, 2025-10-02:
I couldn't do anything with my Gledopto GL-C-002P 5-in-1 after an incorrect firmware update.
Today, after updating Zigbee2MQTT to version 2.6.2, my device is correctly recognized as a Candeo C-ZB-LC20-Dim and displayed with an image.

Fierys, 2025-10-03:
> Don't celebrate too soon

Could you explain what happened here?

My situation is as follows:
I have three Gledopto GL-C-002P 5-in-1 controllers, which I use in Dimmer mode (GL-C-009P).

When the "incorrect firmware" issue surfaced, I flashed one of the controllers. It continued to function, but in Zigbee2MQTT (Z2M), it was displayed without an image and showed as "not supported."

I was planning to restore the firmware using the instructions in this thread, but I hadn't done so yet.

After updating Z2M to version 2.6.2, this same controller received an image and is now displayed as "supported."

[image]

ViktuK359, 2025-10-03:
Hi ))
I have three Gledopto GL-C-002P 5-in-1 controllers, and tofay after update Z2M do it Unsupport C-ZB-LC20v2-RGBCCT.
But device is work well.

IR-Cat, 2026-09-14:
[image]

[image]

Bought a new one today and it asks for update for the same FW as "Candeo". Is the problem still exists?
Both are GL-C-201P. Update on reflashed one just annoys cause it crashes every time and I can't hide it forever but for a new one I just scared to reflash it for a wrong firmware again