Koenkk/zigbee2mqtt #28625: New BSEED roller switch is not exposing calibration, position and backlight options

thogens, 2025-09-14:
### What happened?

Hi,

I have a well working (older) BSEED roller switch in my system and now brought some new BSEED roller switches and added them into z2m.

PROBLEM STATEMENT:
**The new switches are neither exposing the calibration function, nor position function, nor indicator nor are they having backlight options.**
I can't calibrate them at all. (I tried many options I found in internet, but none of them worked. - long press until blinking etc...)

The new switches are recognised as:
Model: TS0601
Manfacturer: _TZE284_uqfph8ah

The old (working switches) are recognised like this:
Model: TS130F
Manfacturer: _TZ3000_vsj5o7dm

SOLUTION IDEA:
---> Does it make sense to force detecting the new ones as "old" models?
If yes, how can I do that?

Is there any other fix?

### What did you expect to happen?

I expect that the roller switches expose all the "old" options that the TS130F have.

### How to reproduce it (minimal and precise)

Pair new roller switches and see that there are no options, other than open / close (for the fixed preset time)

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

20240716

### Adapter

SLZB

### Setup

Using Z2M add-on on HomeAssistant on virtual machine.

Koenkk, 2025-09-14:
Stopping z2m, chnaging the manufacturerName to the old one will trick z2m in using the old cover code

thogens, 2025-09-14:
Where do I need to change that?

thogens, 2025-09-18:
Update on 

> Second issue: The devices are losing their calibration after power cycle. Not sure if this is normal? My old BSEED switch that was recognised as Tuya TS130F does not have this issue.

-> For whatever reason, the devices are no longer losing their calibration. Seems after a night with power and opening and closing events, the calibration persists. - Strange, but works now. Closing the ticket. Thanks.

closed 2025-09-18 by thogens

andersanz93-afk, 2025-09-25:
i'm not able to calibrate the shuttet correctly... any tip? every time i calibrate the shutter it stops midway.... i've been 1 week to try to do it properly but there is no way.... every time stops in the middle

thogens, 2025-09-25:
Hi!

Have you tried the following?
Switch calibration on
Move blinds fully up
Hit close button on the switch (not in z2m)
The blinds should fully close now
Hit stop button once it is fully closed
Switch calibration off to save.

andersanz93-afk, 2025-09-25:
i'm able to fully close the shutter while calibration ins on, but when i end the calibration and rise the sutter, it stays in the middle

andersanz93-afk, 2025-09-25:
okay, i was able to fix it, the issue is that i was trying to set the calibration through the PC, thanks for the advice @thogens!!! i was about to quit.

abcben78, 2025-12-02:
Hello,

I'm not even able to calibrate this switch, good use some help...

Ben

thogens, 2025-12-02:
> Hi!
> 
> Have you tried the following? Switch calibration on Move shutter fully up Hit close button on the switch (not in z2m) The shutter should fully close now Hit stop button once it is fully closed Switch calibration off to save.