Koenkk/zigbee2mqtt #28626: Tuya Knob wrongly identified / duplicated?

terragady, 2025-09-14:
### What happened?

I am setting up new instance freshly and adding devices, I have a Tuya Knob which was discovered as this one https://www.zigbee2mqtt.io/devices/ZG-101Z_D_1.html where I could use step size but now when I add it, it is discovered as https://www.zigbee2mqtt.io/devices/ZG-101Z_D.html which is the same device but with fewer options? 

Is it possible to manualy switch it?

### What did you expect to happen?

To be recognized as https://www.zigbee2mqtt.io/devices/ZG-101Z_D_1.html

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

20250321

### Adapter

zdongleP

### Setup

HA addon

### Device `database.db` entry

{"id":3,"type":"EndDevice","ieeeAddr":"0xa4c13829bd3e5fd6","nwkAddr":25870,"manufId":4742,"manufName":"_TZ3000_gwkzibhs","powerSource":"Battery","modelId":"TS004F","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":260,"inClusterList":[0,1,3,4,6,4096],"outClusterList":[25,10,3,4,5,6,8,4096],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"genBasic":{"attributes":{"manufacturerName":"_TZ3000_gwkzibhs","zclVersion":3,"appVersion":147,"modelId":"TS004F","powerSource":3}},"genOnOff":{"attributes":{"tuyaOperationMode":1}}},"binds":[{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b00258b7685","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":65000,"repChange":10}],"meta":{}}},"appVersion":147,"stackVersion":2,"hwVersion":1,"dateCode":"20250612","swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1757858762243}

terragady, 2025-09-14:
read some other issues and changing appVersion from 147 to 1477 fixed it but now it is discovered as [HOBEIAN](https://www.zigbee2mqtt.io/supported-devices/#v=HOBEIAN) with wrong image.

Koenkk, 2025-10-30:
Fixed in https://github.com/Koenkk/zigbee-herdsman-converters/pull/10562

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-10-30 by Koenkk