Koenkk/zigbee2mqtt #28633: [New device support]: TS0601 by _TZE200_eqpaxqdv KnockautX cover controller

phiphi060, 2025-09-15:
> This issue is stale because it has been open 60 days with no activity. Remove stale label or comment or this will be closed in 7 days 

 _Originally posted by @github-actions in [#25118](https://github.com/Koenkk/zigbee2mqtt/issues/25118#issuecomment-2727068830)_

<html><body>
<div data-testid="markdown-body" data-team-hovercards-enabled="true" class="markdown-body" data-turbolinks="false"><div class="Box-sc-g0xbh4-0 markdown-body NewMarkdownViewer-module__safe-html-box--dKCgP"><p dir="auto">Hi,<br>
Using a Tuya gateway, I was able to obtain the Tuya Data Points:<br>
`</p>

&lt;style&gt;

&lt;/style&gt;

<markdown-accessiblity-table data-catalyst="">
TuyaDatapoints | Type | Tuya Value | Z2mqtt Converter | Ok/Ko
-- | -- | -- | -- | --
1:"Control" | Enum | open/stop/close | state | ok
2:"Curtain   position setting" | Integer | 0…100 scale 0 step 1 | position | ok
3:"Current   curtain position" | Integer | 0…100 scale 0 step 1 | position | ok
4:"Mode" | Enum | ? Markiesen Funktion | opening_mode | ?
5:"Motor  running direction" | Enum | Forward running / Inverted running | motor_direction | ok
6:"Add   wireless remote" | ?Enum | OFF/ON |   |  
7:"Work   State" | Enum | opening/closing | work_state | ?
10:"Total   Time" | Integer | 0... 600 s (17 s) unit "s" | calibration_time | ok
11:"Situation_set" | Enum | fully open/fully closed |   |  
12:"Fault" | Bitmap |   | motor_fault | ?
16:"Lower   limit setting" (chinois) |   |   | set_bottom_limit | ?
19:"Best   Position" | Integer | 1…100 | favorite_position | ok
21:"Angle   Control" | Integer | 0/25/50/75/100° |   | ok state ko set
101:"travel   calibration" |   | Start | calibration | ok
102:"Quick   Calibration 1" |   | Start |   | ok
103:"Best   position" | Integer | OFF/ON | goto_position | ko
104:"Reset" | ?Enum |   | factory_reset | ?

</markdown-accessiblity-table>

`
Based on the messages above, I created an external converter:
<p dir="auto">`const {} = require('zigbee-herdsman-converters/lib/modernExtend');<br>
// Add the lines below<br>
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');<br>
const tz = require('zigbee-herdsman-converters/converters/toZigbee');<br>
const exposes = require('zigbee-herdsman-converters/lib/exposes');<br>
const reporting = require('zigbee-herdsman-converters/lib/reporting');<br>
const ota = require('zigbee-herdsman-converters/lib/ota');<br>
const tuya = require('zigbee-herdsman-converters/lib/tuya');<br>
const {} = require('zigbee-herdsman-converters/lib/tuya');<br>
const utils = require('zigbee-herdsman-converters/lib/utils');<br>
const globalStore = require('zigbee-herdsman-converters/lib/store');<br>
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');<br>
const e = exposes.presets;<br>
const ea = exposes.access;</p>
<p dir="auto">const definition = {<br>
fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_eqpaxqdv' },],<br>
zigbeeModel: ['TS0601'],<br>
model: 'TS0601',<br>
vendor: '_TZE200_eqpaxqdv',<br>
description: 'KnockautX Multi STAK 3 / STAS 3',<br>
fromZigbee: [tuya.fz.datapoints],<br>
toZigbee: [tuya.tz.datapoints],<br>
exposes: [<br>
e.cover_position().setAccess('position', ea.STATE_SET),<br>
e.enum('opening_mode', ea.STATE_SET, ['tilt', 'lift']).withDescription('Switch opening mode').withCategory('config'),<br>
e.enum('motor_direction', ea.STATE_SET, ['normal', 'reversed']).withDescription('Set the motor running direction'),<br>
e.enum('add_wireless_remote', ea.STATE_SET, ['START', 'END']).withDescription('Add a wireless remote'),<br>
e.text('work_state', ea.STATE, ['opening', 'closing']),<br>
e.numeric('calibration_time', ea.ALL).withValueMin(0).withValueMax(600).withUnit('s'),<br>
e.text('situation_set', ea.STATE, ['fully_open', 'fully_close']),<br>
e.binary('motor_fault', ea.STATE, true, false).withDescription('Motor Fault'),<br>
e.enum('set_bottom_limit', ea.STATE_SET, ['START', 
'END']).withDescription('Set the bottom limit, to reset limits use 
factory_reset'),<br>
e.binary('goto_positon', ea.SET, true, false),<br>
e.enum('angle_control', ea.STATE_SET, ['0', '25', '50', '75', '100']),<br>
e.enum('calibration', ea.STATE_SET, ['START', 'END']).withDescription('Travel Calibration'),<br>
//e.enum('calibration1', ea.STATE_SET, ['START', 'END']).withDescription('Quick Calibration'),<br>
e.numeric('favorite_position', 
ea.STATE_SET).withValueMin(1).withValueMax(100).withDescription('Favorite
 position of this cover'),//error with ValueMin(0)<br>
e.binary('factory_reset', ea.SET, true, false).withDescription('Factory reset the device'),<br>
],<br>
whiteLabel: [<br>
tuya.whitelabel('KnockautX', 'PIMS3028', 'Cover', ['_TZE200_eqpaxqdv'])<br>
],<br>
configure: tuya.configureMagicPacket,<br>
meta: {<br>
tuyaDatapoints: [<br>
[1, 'state', tuya.valueConverterBasic.lookup({OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2)})],<br>
[2, 'position', tuya.valueConverter.coverPosition],//Curtain position setting<br>
[3, 'position', tuya.valueConverter.coverPositionInverted],//Current curtain position<br>
[4, 'opening_mode', tuya.valueConverterBasic.lookup({tilt: tuya.enum(0), lift: tuya.enum(1)})], // no effect<br>
[5, 'motor_direction', tuya.valueConverter.tubularMotorDirection],<br>
[6, 'add_wireless_remote', tuya.valueConverterBasic.lookup({START: tuya.enum(0), END: tuya.enum(1)})],<br>
[7, 'work_state', tuya.valueConverterBasic.lookup({opening: tuya.enum(0), closing: tuya.enum(1)})], // don't work<br>
[10, 'calibration_time', tuya.valueConverter.raw],//17s for my cover, default 600s<br>
[11, 'situation_set', tuya.valueConverterBasic.lookup({fully_open: tuya.enum(0), fully_close: tuya.enum(1)})],//don't work<br>
[16, 'set_bottom_limit', tuya.valueConverterBasic.lookup({START: tuya.enum(1), END: tuya.enum(0)})],//don't work, error<br>
[19, 'favorite_position', tuya.valueConverter.setLimit],<br>
[21, 'angle_control', tuya.valueConverterBasic.lookup({ 0: tuya.enum(0),
 25: tuya.enum(25), 50: tuya.enum(50), 75: tuya.enum(75), 
100:tuya.enum(100)})],//State work, Set don't work<br>
[101, 'calibration', tuya.valueConverterBasic.lookup({START: tuya.enum(0), END: tuya.enum(1)})], // travel calibration<br>
//[102, 'calibration1', tuya.valueConverterBasic.lookup({START: tuya.enum(0), END: tuya.enum(1)})], // Quick Calibration 1<br>
[103, 'goto_position', tuya.valueConverter.raw],//don't work<br>
[104, 'factory_reset', tuya.valueConverter.raw],//no effect ?<br>
],<br>
},<br>
};</p>
<p dir="auto">module.exports = definition;`</p>
<p dir="auto">You can see in the comments what works and what doesn't.<br>
Attached is another picture of the device</p>
<a target="_blank" rel="noopener noreferrer" href="https://private-user-images.githubusercontent.com/166981887/489461419-8df5546b-22ac-4caa-a6a0-6c36e0fba496.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc5MzEyMzQsIm5iZiI6MTc1NzkzMDkzNCwicGF0aCI6Ii8xNjY5ODE4ODcvNDg5NDYxNDE5LThkZjU1NDZiLTIyYWMtNGNhYS1hNmEwLTZjMzZlMGZiYTQ5Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkxNVQxMDA4NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZTkyYzA2ZGZkNmQyMWFmOWMzODhmNGY5YTQ5M2UzY2FjZjM1ZjQyMTYwODYwOTY3Y2MzYTQ4OTkxODIyZTkzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.H69PnS6hh_fm-Shr0pi6vcdI2OT3LW3-xzLxdOUUlXc"><img width="512" height="512" alt="Image" src="https://private-user-images.githubusercontent.com/166981887/489461419-8df5546b-22ac-4caa-a6a0-6c36e0fba496.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTc5MzEyMzQsIm5iZiI6MTc1NzkzMDkzNCwicGF0aCI6Ii8xNjY5ODE4ODcvNDg5NDYxNDE5LThkZjU1NDZiLTIyYWMtNGNhYS1hNmEwLTZjMzZlMGZiYTQ5Ni5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwOTE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDkxNVQxMDA4NTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZTkyYzA2ZGZkNmQyMWFmOWMzODhmNGY5YTQ5M2UzY2FjZjM1ZjQyMTYwODYwOTY3Y2MzYTQ4OTkxODIyZTkzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.H69PnS6hh_fm-Shr0pi6vcdI2OT3LW3-xzLxdOUUlXc" style="max-width: 100%; height: auto; max-height: 512px;"></a></div></div>
</body>
</html>

blehers, 2025-11-17:
@phiphi060 Today the manufacturer sent me a complete list of all data points. It would be brilliant if you could complete the external converter and create a pull request for everyone to use.

| DP ID | DP Name                  | Identifier                       | Data Transfer Type | Data Type | Properties                                 |
| ----- | ------------------------ | -------------------------------- | ------------------ | --------- | ------------------------------------------ |
| 1     | Control                  | control                          | Send and Report    | enum      | open, stop, close, lock, unlock            |
| 2     | Curtain position setting | percent_control                  | Send and Report    | value     | range: 0-100; pitch: 1; scale: 0; unit:    |
| 3     | Current curtain position | percent_state                    | Report Only        | value     | range: 0-100; pitch: 1; scale: 0; unit:    |
| 4     | Mode                     | mode                             | Send and Report    | enum      | up, up_delete, remove_up_down              |
| 5     | Motor running direction  | control_back / control_back_mode | Send and Report    | enum      | forward, back                              |
| 6     | Add wireless remote      | auto_power                       | Send and Report    | bool      |                                            |
| 7     | Work State               | work_state                       | Report Only        | enum      | opening, closing, 123                      |
| 10    | Total Time               | time_total                       | Report Only        | value     | range: 0-600; pitch: 1; scale: 0; unit: s  |
| 11    | Situation_set            | situation_set                    | Report Only        | enum      | fully_open, fully_close                    |
| 12    | Fault                    | fault                            | Report Only        | fault     | motor_fault                                |
| 16    | border                   | border                           | Send and Report    | enum      | down_delete, remove_top_bottom             |
| 19    | Best Position            | position_best                    | Send and Report    | value     | range: 0-100; pitch: 1; scale: 0; unit:    |
| 21    | Angle Control            | angle_horizontal                 | Send and Report    | value     | range: 0-100; pitch: 25; scale: 0; unit: ° |
| 101   | travel calibration       | cur_calibration                  | Send and Report    | enum      | start, end                                 |
| 102   | Quick Calibration 1      | quick_calibration_1              | Send and Report    | value     | range: 0-900; pitch: 1; scale: 0; unit: s  |
| 103   | Best position            | switch                           | Send and Report    | bool      |                                            |
| 104   | Reset                    | reset                            | Send and Report    | enum      | reset                                      |

blehers, 2025-11-17:
If you don't already know, you can also find a detailed [user manual](https://drive.google.com/drive/folders/17pOKjynKYwis_MZMXs8nCdW-vj1LH35K) (in German) on their [website](https://www.brelag.com/shop/pims3028-knockautx-plugin-empfanger-multi-stak-3-stas-3-4517?category=6#nav_tabs_content_1646678141574_458).

blehers, 2025-12-06:
@phiphi060 @nabossha

Could anyone test? My blinds aren't installed yet.

````javascript
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
	fingerprint: [{ modelID: 'TS0601', manufacturerName: '_TZE200_eqpaxqdv' }],
	zigbeeModel: ['TS0601'],
	model: 'TS0601',
	vendor: '_TZE200_eqpaxqdv',
	description: 'KnockautX Multi STAK 3 / STAS 3',
	fromZigbee: [tuya.fz.datapoints],
	toZigbee: [tuya.tz.datapoints],
	exposes: [
		// DP 1: control; lock/unlock to protect the blinds in stormy weather
		e.enum('control', ea.STATE_SET, ['open', 'stop', 'close', 'lock', 'unlock'])
			.withDescription('Control commands'),
		// DP 2/3: curtain position
        e.cover_position().setAccess('position', ea.STATE_SET),
		// DP 4: mode (up, up_delete, remove_up_down); probably used for configuring the motor limits
		e.enum('mode', ea.STATE_SET, ['up', 'up_delete', 'remove_up_down'])
			.withDescription('Set mode'),
		// DP 5: motor runnning direction
		e.enum('control_back', ea.STATE_SET, ['forward', 'back'])
			.withDescription('Set the motor running direction'),
		// DP 6: Add wireless remote
		e.binary('auto_power', ea.STATE_SET, true, false)
            .withDescription('Add wireless remote'),
		// DP 7: Work state (report only)
        e.enum('work_state', ea.STATE, ['opening', 'closing', '123'])
            .withDescription('Work state'),
		// DP 10: Total Time (report only)
        e.numeric('time_total', ea.STATE).withValueMin(0).withValueMax(600).withUnit('s')
            .withDescription('Total travel time'),
		// DP 11: Situation set (report only)
		e.enum('situation_set', ea.STATE, ['fully_open', 'fully_close'])
            .withDescription('Situation of the blinds'),
		// DP 12: Motor fault (report only)
		e.text('fault', ea.STATE)
			.withDescription('Motor fault (raw)'),
		// DP 16: Border
		e.enum('border', ea.STATE_SET, ['down_delete', 'remove_top_bottom'])
			.withDescription('Set lower/upper limit'),
        // DP 19: Best position
        e.numeric('position_best', ea.STATE_SET).withValueMin(1).withValueMax(100) // error withValueMin(0)?
            .withDescription('Set the best position'),
        // DP 21: Angle control (numeric 0/25/50/75/100)
		e.numeric('angle_horizontal', ea.STATE_SET).withValueMin(0).withValueMax(100).withValueStep(25).withUnit('°')
            .withDescription('Set the angle'),
		// DP 101: Travel calibration
        e.enum('cur_calibration', ea.STATE_SET, ['start', 'end'])
            .withDescription('Calibrate the travel limits'),
        // DP 102: Quick calibration
        e.numeric('quick_calibration_1', ea.STATE_SET).withValueMin(0).withValueMax(900).withUnit('s')
            .withDescription('Set quick calibration'),
        // DP 103: Best position
		e.binary('switch', ea.STATE_SET, true, false)
            .withDescription('Trigger best position'),
		// DP 104: Reset
		e.enum('reset', ea.STATE_SET, ['reset'])
            .withDescription('Trigger factory reset'),
	],
	whiteLabel: [
		tuya.whitelabel('KnockautX', 'PIMS3028', 'Cover', ['_TZE200_eqpaxqdv']),
	],
	configure: tuya.configureMagicPacket,
		meta: {
			tuyaDatapoints: [
				// 1: Control
				[1, 'control', tuya.valueConverterBasic.lookup({ open: tuya.enum(0), stop: tuya.enum(1), close: tuya.enum(2), lock: tuya.enum(3), unlock: tuya.enum(4) })],
				// 2: Curtain position setting (set) -> standard `position` instead of percent_control
				[2, 'position', tuya.valueConverter.coverPosition],
				// 3: Current curtain position (report) -> standard `position` instead of percent_state
				[3, 'position', tuya.valueConverter.coverPosition],
				// 4: Mode
				[4, 'mode', tuya.valueConverterBasic.lookup({ up: tuya.enum(0), up_delete: tuya.enum(1), remove_up_down: tuya.enum(2) })],
				// 5: Motor running direction / control_back
				[5, 'control_back', tuya.valueConverterBasic.lookup({ forward: tuya.enum(0), back: tuya.enum(1) })],
				// 6: Add wireless remote / auto power
				[6, 'auto_power', tuya.valueConverterBasic.lookup({ 0: tuya.enum(0), 1: tuya.enum(1) })],
				// 7: Work state (report only)
				[7, 'work_state', tuya.valueConverterBasic.lookup({ opening: tuya.enum(0), closing: tuya.enum(1), 123: tuya.enum(2) })],
				// 10: Total time
				[10, 'time_total', tuya.valueConverter.raw],
				// 11: Situation set
				[11, 'situation_set', tuya.valueConverterBasic.lookup({ fully_open: tuya.enum(0), fully_close: tuya.enum(1) })],
				// 12: Fault / motor fault
				[12, 'fault', tuya.valueConverter.raw],
				// 16: Border / lower limit setting
				[16, 'border', tuya.valueConverterBasic.lookup({ down_delete: tuya.enum(0), remove_top_bottom: tuya.enum(1) })],
				// 19: Best position
				[19, 'position_best', tuya.valueConverter.setLimit],
				// 21: Angle control / horizontal angle (numeric)
				[21, 'angle_horizontal', tuya.valueConverter.raw],
				// 101: Travel calibration
				[101, 'cur_calibration', tuya.valueConverterBasic.lookup({ start: tuya.enum(0), end: tuya.enum(1) })],
				// 102: Quick calibration (seconds)
				[102, 'quick_calibration_1', tuya.valueConverter.raw],
				// 103: Best position switch (boolean 0/1)
				[103, 'switch', tuya.valueConverterBasic.lookup({ 0: tuya.enum(0), 1: tuya.enum(1) })],
				// 104: Reset
				[104, 'reset', tuya.valueConverterBasic.lookup({ reset: tuya.enum(0) })],
			],
		},
};

nabossha, 2025-12-07:
@blehers looked at it again today, I had to make 2 changes on the DataPoints to make my blinds work and report correctly with HomeAssistant:

`````
// 1: Control
[1, 'state', tuya.valueConverterBasic.lookup({OPEN: tuya.enum(0), STOP: tuya.enum(1), CLOSE: tuya.enum(2)})],

// 3: Current curtain position (report) -> standard `position` instead of percent_state
[3, 'position', tuya.valueConverter.coverPositionInverted],
````

phiphi060, 2025-12-09:
Here the link to my Fork [https://github.com/phiphi060/zigbee-herdsman-converters/blob/patch-1/src/devices/tuya.ts](url)