Koenkk/zigbee2mqtt #28635: So much things going wrong - multiple errors and unavailable devices / unresponsive UI

timwnl, 2025-09-15:
### What happened?

Suddenly my Zigbee2MQTT instance is starting to trow errors and certain devices do no longer turn online. I experience when I press the join button, nothing happens (but I do see in the logs the joining has started). Basically parts of the devices are not available, while the UI isn't working as expected too. 

z2m: Failed to read state of 'Achtertuin Airco Plug Tuya' after reconnect (ZCL command 0x70ac08fffe529e7e/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - UTIL - assocGetWithAddress after 6000ms))

z2m: Failed to ping 'Keuken Keukenspot Koffiemachine Hue' (attempt 1/1, ZCL command 0x00178801096c810c/11 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (SRSP - AF - dataRequest after 6000ms))

z2m: Failed to read state of 'Achtertuin Navimow Plug' after reconnect (ZCL command 0xf0d1b8be240b4990/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 17494 - 1 - 98 - 6 - 1 after 10000ms))

z2m: Failed to read state of 'Slaapkamer Spot 2 Hue' after reconnect (ZCL command 0x0017880108cc9840/11 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))

Some devices which are causing errors (they are always the same) I force remove them from the UI. One device I am unable to re-join and get interview challenges. 

zh:controller: Interview failed for '0xa4c138b115eb3649 with error 'Error: Interview failed because can not get active endpoints ('0xa4c138b115eb3649')'

I have two Sonoff ZBDongle-P - I have reflashed them with the latest coordinator firmwares. Using https://dongle.sonoff.tech/sonoff-dongle-flasher/ and the CC1352P2_CC2652P_launchpad_coordinator_20250321.hex. I have cloned the IEEE with the second Sonoff ZBDongle-P with ZigStar. I do assume I have ruled out an issue with the HW (in this case).

I tried 'latest', 'latest-dev', '2.6.1' and '2.6.0'. They all give the same issues as recorded above. I also removed and rebuild the docker images. 

### What did you expect to happen?

I am actually getting lost in the amount of errors and I can't find the 'root cause' issues. My expectations is that all devices are reporting back online and are responsive. 

### How to reproduce it (minimal and precise)

I have no clue what is going wrong. Hope somebody can help / provide guidance.

### Zigbee2MQTT version

latest-dev, latest, 2.6.1, 2.6.0

### Adapter firmware version

20240710

### Adapter

Sonoff Dongle P (ZStack3x0)

### Setup

I run Zigbee2MQTT in docker, on a synology DS720+. It worked for years, no significant updates performed in the last couple of days. Errors came out of nowhere

### Debug log

[log.log](https://github.com/user-attachments/files/22335345/log.log)

peletonduwer, 2025-09-17:
Same here, tons of errors from two identical devices since the last ZigBee2MQTT update. Hopefully, it'll be fixed in the next update.

pypb, 2025-09-19:
From the looks of it, same thing happened on my setup two days ago. Most devices offline, lots of log entries like these:

```
[2025-09-17 19:54:19] error:    zh:controller:device: Read response to 0xa4c1387c337baf80 failed (ZCL command 0xa4c1387c337baf80/1 genTime.readRsp({"localTime":811454060}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":215,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (241)))
[2025-09-17 19:54:20] error:    zh:controller:device: Read response to 0xa4c1387c337baf80 failed (ZCL command 0xa4c1387c337baf80/1 genTime.readRsp({"localTime":811454060}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":215,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (241)))
[2025-09-17 19:54:20] error:    zh:controller:device: Read response to 0xa4c1387c337baf80 failed (ZCL command 0xa4c1387c337baf80/1 genTime.readRsp({"localTime":811454060}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":215,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (241)))
[2025-09-17 20:02:12] warning:  z2m: Failed to ping 'living_room_wall_dimmer' (attempt 1/2, ZCL command 0xd0cf5efffe49147a/3 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'undefined' (241)))
```

Restarting z2m did not help. I had to unplug/replug the stick to solve the problem.

I have a Electrolama zzh stick running Z-Stack 20250321 firmware. I recently upgraded to 20250321, previously I was running 20230507 which was completely stable. If this happens again, I might downgrade to see if that solves the issue.

Ley21, 2025-09-25:
I have the same issues since today morning with Z2M (2.6.1-1) + SLZB-06 (15/05/2025) and Core (2025.9.4). Restart of Z2M or SZLB are not helping at all, all my devices are offline and I dont know how to get them online again. If anyone has a idea please let me know. (Reseting is not a solutions, since some of the devices are hardwired).

pypb, 2025-11-22:
Just thought I’d check back. It’s been over two months with no issues, so whatever caused the problem seems to have been a one-time thing. 🤞

fbloemhof, 2025-12-19:
I also experience this issue using the SLZB-MR3 stick.

I contacted Smlight about it and they say the Zigbee CC26XX chip is rhe issue and a fix from Texas Instruments is needed? So maybe it is an firmware issue then?