Koenkk/zigbee2mqtt #28639: SMLight SLZB-06Mg24 ethernet coordinator crashing and some bulbs leaving network

apydo, 2025-09-15:
### What happened?

I migrate my current network from USB coordinator to Ethernet one : Sonoff Dongle E -> SMLight SLZB-06Mg24.
It wasn't very smooth and I had some problems, however I didn't have to repair ALL the devices.
Things seemed to work fine for some times.
But I have some power outages and when it happens sometimes Z2M has some difficulties to restart. 2 or 3 restarts are sometimes necessary for Z2M to survive.
Moreover when Z2M restarts some devices leaves the network. It happens mainly with IKEA bulbs and sometimes with Philips bulbs.
These problematic devices are sometimes in Z2M GUI and sometimes some are missing.
If a device left the network I have to reset it to join the network again. This doesn't always work. I have to open network on a specific router to have device join the network again. It's possible that the coordinator is overloaded because there is a lot of devices in this network (75).

### What did you expect to happen?

I'd like to have some advices to avoid Z2M crashes and devices leaving network.
Having Z2M stopped and started without crashes and devices leaving network would be great.

### How to reproduce it (minimal and precise)

Stop Z2M for some time, maybe stop the Ethernet adapter/coordinator as well for some time. Restart again.

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

Ember 8.2.1

### Adapter

SLZB-06Mg24, ethernet mode

### Setup

Plain Z2M in a LXC container on Proxmox VE 9 Cluster (x86-64).

# uname -a
Linux zigbee2mqtt1 6.14.11-2-pve #1 SMP PREEMPT_DYNAMIC PMX 6.14.11-2 (2025-09-12T09:46Z) x86_64 GNU/Linux
2 cpus, RAM: 1024 MB
Node.js version : 24.7.0

28 routers
47 end devices

### Debug log

Don't have any debug log at this time but I can see errors such as : 

[2025-09-09 13:05:54] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_ADDRESS_CONFLICT for "0".

Or

[2025-09-15 14:27:37] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "40834".

Or

[2025-09-14 16:55:30] info:     zh:ember:ezsp: Received network/route error ROUTE_ERROR_NON_TREE_LINK_FAILURE for "35689".

Or

[2025-09-09 13:05:55] warning:  zh:ember:uart:ash: Frame(s) in progress cancelled in [1ac2020682af7e]
[2025-09-09 13:05:55] error:    zh:ember:uart:ash: Received ERROR from adapter, with code=RESET_ASSERT.
[2025-09-09 13:05:55] error:    zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
[2025-09-09 13:05:55] error:    zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
[2025-09-09 13:05:55] error:    zh:ember: Adapter fatal error: HOST_FATAL_ERROR
[2025-09-09 13:05:55] info:     zh:ember:uart:ash: ASH COUNTERS since last clear:
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Total frames: RX=2401, TX=2776
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Cancelled   : RX=1, TX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   DATA frames : RX=2399, TX=375
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   DATA bytes  : RX=48364, TX=9377
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Retry frames: RX=0, TX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   ACK frames  : RX=0, TX=2400
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   NAK frames  : RX=0, TX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   nRdy frames : RX=0, TX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   CRC errors      : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Comm errors     : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Length < minimum: RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Length > maximum: RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Bad controls    : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Bad lengths     : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Bad ACK numbers : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Out of buffers  : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Retry dupes     : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   Out of sequence : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash:   ACK timeouts    : RX=0
[2025-09-09 13:05:55] info:     zh:ember:uart:ash: ======== ASH stopped ========
[2025-09-09 13:05:55] info:     zh:ember:ezsp: ======== EZSP stopped ========
[2025-09-09 13:05:55] info:     zh:ember: ======== Ember Adapter Stopped ========
[2025-09-09 13:05:55] error:    z2m: Adapter disconnected, stopping
[2025-09-09 13:05:55] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"offline"}'
[2025-09-09 13:05:55] info:     z2m: Disconnecting from MQTT server
[2025-09-09 13:05:55] info:     z2m: Stopping zigbee-herdsman...
[2025-09-09 13:05:55] info:     z2m: Stopped zigbee-herdsman
[2025-09-09 13:05:55] info:     z2m: Stopped Zigbee2MQTT

Nerivec, 2025-09-22:
The missed initial IEEE change might be a problem here. Could be some devices are now trying to find the wrong coordinator when they get into specific states.
If you re-pair one of these devices that drops off, will it drop off again, or is it then fixed? (make sure to do a long-press reset on the device, which should hopefully factory-reset it)
If something has gone wrong in the adapter with the network data, you may end up having to re-pair everything unfortunately...
As a last resort, you could try [backing up the tokens](https://github.com/Nerivec/ember-zli/wiki/Stack#backup-nvm3-tokens) from your old adapter (assuming you have not reset it yet), and restoring them back onto the new one (that could create some counters issues since you likely have gone through quite a bit of messaging since, but last resort before doing a full re-pair...).

_If there is no massive undertaking in re-pairing the whole network (like having to open walls or something), I'd suggest full re-pairing right away, you likely will lose as much time in trying to recover, and a clean re-pair is the only way to get things back to a known state. You can [factory-reset the adapter in ember-zli bootloader](https://github.com/Nerivec/ember-zli/wiki/Bootloader#clear-nvm3)._

apydo, 2025-09-22:
> If you re-pair one of these devices that drops off, will it drop off again, or is it then fixed?

Strangely, the only devices that leaves the network or stop working are the bulbs.
Most of these are from IKEA (Tradfri) or Philips (2 of them).
I have to factory reset to re-pair.
Everything else seems to work fine.

The old adapter is not used for a long time and I added some devices to the network. I'm afraid it's too late now to recover.

> _If there is no massive undertaking in re-pairing the whole network (like having to open walls or something), I'd suggest full re-pairing right away, you likely will lose as much time in trying to [...]

Well, I'll do that.
Do you know what I have to do to simplify re-pairing ?
First, you suggest to factory-reset the adapter. Is there any possibility to connect to the ethernet adapter with ember-zli ?
I'd like to preserve devices names in Z2M and automations in HA. Maybe I can disable mqtt in Z2M config to avoid any interaction with HA during re-pairing ?
I suppose I have to reset something in Z2M as well ? I can even create a new LXC container in my Proxmox cluster to start from scratch. The only concern here is to get back all the devices names so I don't have to rename each device one by one.

Your tips are welcome.

apydo, 2025-09-30:
I close this issue as I can't see any problem anymore. Thanks to @Nerivec for his valuable help.

closed 2025-09-30 by apydo