Koenkk/zigbee2mqtt #28650: Samotech SM325-ZG query ("polling" frequency) - Received ZDO response

Great-Chart, 2025-09-16:
### What happened?

Not so much an issue per-se but I have been using the Health feature within new Z2M front end to look out for overly chatty devices.  This has identified some that I have removed for test purposes (where they might have had potential to be flooding the network).

One particular device continues to appear in the logs on a 60 second cycle
https://www.zigbee2mqtt.io/devices/SM325-ZG.html

It is quite limited in it's features and just exposes state and brightness.
The log entries for this device are unique for the remainder of my devices in that it references Received ZDO response that I'm uncertain if this is indicating an issue or is perfectly normal.

**Anyone able to advise here?
Ditto anyone with this corded dimmer able to advise if their unit is updating every 60 seconds?
Is there any way to alter the behaviour of this device so it updates on a more normal basis (ie 10 mins etc)?**

```
[2025-09-11 08:13:33] [34mdebug[39m: 	zh:controller: Received ZDO response: clusterId=NETWORK_ADDRESS_RESPONSE, status=SUCCESS, payload={"eui64":"0xf4b3b1fffe63bc0b","nwkAddress":1022,"startIndex":0,"assocDevList":[]}
[2025-09-11 08:13:33] [34mdebug[39m: 	zh:controller: Network address from '0xf4b3b1fffe63bc0b:1022'
[2025-09-11 08:13:33] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Light', payload '{"brightness":254,"last_seen":"2025-09-11T08:13:33+01:00","linkquality":87,"state":"OFF"}'

[2025-09-11 08:14:32] [34mdebug[39m: 	zh:controller: Received ZDO response: clusterId=NETWORK_ADDRESS_RESPONSE, status=SUCCESS, payload={"eui64":"0xf4b3b1fffe63bc0b","nwkAddress":1022,"startIndex":0,"assocDevList":[]}
[2025-09-11 08:14:32] [34mdebug[39m: 	zh:controller: Network address from '0xf4b3b1fffe63bc0b:1022'
[2025-09-11 08:14:32] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Light', payload '{"brightness":254,"last_seen":"2025-09-11T08:14:32+01:00","linkquality":87,"state":"OFF"}'

[2025-09-11 08:15:32] [34mdebug[39m: 	zh:controller: Received ZDO response: clusterId=NETWORK_ADDRESS_RESPONSE, status=SUCCESS, payload={"eui64":"0xf4b3b1fffe63bc0b","nwkAddress":1022,"startIndex":0,"assocDevList":[]}
[2025-09-11 08:15:32] [34mdebug[39m: 	zh:controller: Network address from '0xf4b3b1fffe63bc0b:1022'
[2025-09-11 08:15:32] [32minfo[39m: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Bathroom Light', payload '{"brightness":254,"last_seen":"2025-09-11T08:15:32+01:00","linkquality":87,"state":"OFF"}'
```

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle-P

### Setup

AMD A4 PRO-7350B

### Device `database.db` entry

{"id":18,"type":"Router","ieeeAddr":"0xf4b3b1fffe63bc0b","nwkAddr":1022,"manufId":4107,"manufName":"Samotech","powerSource":"Mains (single phase)","modelId":"SM325-ZG","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"appVersion":0,"stackVersion":0,"hwVersion":0,"zclVersion":8}},"genOnOff":{"attributes":{"onOff":0}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc1862","endpointID":1},{"cluster":8,"type":"endpoint","deviceIeeeAddress":"0x00124b0024bc1862","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1},{"cluster":8,"attrId":0,"minRepIntval":5,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":0,"stackVersion":0,"hwVersion":0,"dateCode":"20231206","swBuildId":"1.08","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1758037617680}

Koenkk, 2025-09-17:
Could you provide the debug log of this? I expect that the firmware of the device just send this every 60 second, but I have to get the log to make sure it's not z2m that is requesting it.

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Great-Chart, 2025-09-18:
Many thanks Koen,

[log2.log](https://github.com/user-attachments/files/22402424/log2.log)
Log covering the period for this morning; please advise if you want to see the "from start-up" period and I'll restart and pull those.
_Hopefully this is also the correct log too_

Koenkk, 2025-09-22:
I checked the log but z2m doesn't poll it, the device sends the message itself.

> "A device does not have a default reporting interval, all reporting intervals are configured from z2m"

For ZDO NETWORK_ADDRESS_RESPONSE, this is not a type of message for which a reporting interval can be configured for.

closed 2025-11-29 by github-actions[bot]