Koenkk/zigbee2mqtt #28652: [New device support]: New Fingerprint for ZG-103Z, currently detected as TS0210 with no functionality

NeoMod, 2025-09-16:
### Link

https://www.zigbee2mqtt.io/devices/ZG-103Z.html

### Database entry

{"id":36,"type":"EndDevice","ieeeAddr":"0xa4c138155119e4ef","nwkAddr":37363,"manufId":4417,"manufName":"_TZ32101000000_5oy7cysk","powerSource":"Battery","modelId":"TS0210","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,57344,3,1,1280,57346,61184],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"TS0210","manufacturerName":"_TZ32101000000_5oy7cysk","powerSource":3,"zclVersion":3,"appVersion":68,"stackVersion":1,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":68,"stackVersion":1,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1758041594793}

### Zigbee2MQTT version

2.6.1

### Comments

The device is a ZG-103Z, but it is identified as a "TS0210" due to a missing fingerprint (thanks to @ShadowFist for the [insight](https://community.home-assistant.io/t/vibration-sensor-not-showing-any-data/927990/14))

### External definition

```shell
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TS0210'],
    model: 'TS0210',
    vendor: '_TZ32101000000_5oy7cysk',
    description: 'Automatically generated definition',
    extend: [m.battery(), m.iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
    meta: {},
};
```

### What does/doesn't work with the external definition?

The device has the following Zigbee Manufacturer: `_TZ32101000000_5oy7cysk` 
But it is identified as a "TS0210" instead of a ZG-103Z.

ckocyigit, 2025-10-01:
Oh, that's obviously more sensible. I've done it the wrong way around – it should be: https://github.com/ckocyigit/zigbee-herdsman-converters/commit/1b91a0fcf26428689bb396bed10d2d23bd782ff7

zanemendoza123, 2025-10-03:
> Oh, that's obviously more sensible. I've done it the wrong way around – it should be: [ckocyigit/zigbee-herdsman-converters@1b91a0f](https://github.com/ckocyigit/zigbee-herdsman-converters/commit/1b91a0fcf26428689bb396bed10d2d23bd782ff7) right?

I got the same device now and noticed the same issue. And yeah, I think this should work.

fabicodes, 2025-10-03:
The changes unfortunately do not work. Also tested using an external converter on this base; doesn't work yet.

Found out the DP-Ids though
| Name | ID |
| ------ | --- |
 | Shake |  1 |
 | Battery |  4 |
 | X  | 101 |
 | Sensitivity  | 6 |
 | Y |  102 |
 | Z  | 103 |
 | State  | 104 |
 | Sensitivity  | 105 |

I don't know the difference between those 2 sensitivities yet

Those are 2 images from the tuya dashboard

[image]

[image]

pnaklicki, 2025-10-16:
Hey, I am also experiencing this problem with this particular sensor, hope it gets resolved soon!

zanemendoza123, 2025-10-17:
So I got the vibration detection working, but not sure how accurate is it. 
```javascript
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ["TS0210"],
    fingerprint: tuya.fingerprint("TS0210", ["_TZ32101000000_5oy7cysk"]),
    model: "TS0210",
    vendor: "Tuya",
    description: "Vibration sensor",
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.vibration(),
        e.numeric("x", ea.STATE).withValueMin(0).withValueMax(256).withValueStep(1).withDescription("X coordinate"),
        e.numeric("y", ea.STATE).withValueMin(0).withValueMax(256).withValueStep(1).withDescription("Y coordinate"),
        e.numeric("z", ea.STATE).withValueMin(0).withValueMax(256).withValueStep(1).withDescription("Z coordinate"),
        e.battery(),
        e.enum("sensitivity", ea.STATE_SET, ["low", "middle", "high"]).withDescription("Vibration detection sensitivity"),
    ],
    meta: {
        tuyaDatapoints: [
            [4, "battery", tuya.valueConverterBasic],
            [101, "x", tuya.valueConverter.raw],
            [102, "y", tuya.valueConverter.raw],
            [103, "z", tuya.valueConverter.raw],
            [104, "vibration", tuya.valueConverter.trueFalseEnum1],
            [
                105,
                "sensitivity",
                tuya.valueConverterBasic.lookup({
                    low: tuya.enum(0),
                    middle: tuya.enum(1),
                    high: tuya.enum(2),
                }),
            ],
        ],
    },
};

module.exports = definition;
```

jimmyeao, 2025-10-24:
How would I add this to Z2m running as an add on in HA?

BrunoKrugel, 2025-11-22:
anyone got it to work?

NeoMod, 2025-11-24:
> anyone got it to work?

So far, all my attempts have been unsuccessful with this specific model.

marcelfarres, 2025-12-05:
Thank you very much for the clear instruction, it help and now is working no problem :) 

> I can also confirm it is working. It just took me a while do figure out what exactly had to be done. Maybe the following screenshot helps:
> <img alt="Image" width="1013" height="650" src="https://private-user-images.githubusercontent.com/4719666/521237957-f32ca658-552a-4348-be16-026df519e3a0.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NjQ5Njg1OTMsIm5iZiI6MTc2NDk2ODI5MywicGF0aCI6Ii80NzE5NjY2LzUyMTIzNzk1Ny1mMzJjYTY1OC01NTJhLTQzNDgtYmUxNi0wMjZkZjUxOWUzYTAucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MTIwNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTEyMDVUMjA1ODEzWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NjY4OGRkMDgxMzhlMDgwNzgzOGVhYzc1MmY3OTc1NGJlMTQxM2RhYTdiZGU5NDQ2M2EwMmU1ZTRkMGQxYjVhMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ojfKAv5OnTY1siGl4d2C_Wy-pQLE9MZ95qtWZ98pzv0">
> 
> After creating the external converter, remove the device from z2m, reboot (not sure if necessary) and start a new interview.
> 
> After that everything was working smoothly, including home assistant.
> 
> I am so glad that after month of fiddling around with those devices, I can finally finish my mouse trap setup! Thanks for all your help.

mbbelgium, 2025-12-06:
I get this message when I try to add the converter: 

[image]