Koenkk/zigbee2mqtt #28656: [New device support]: Moes Star Feather EU (TS0601, 1, 2, 3, 4-gang).

mattplate, 2025-09-16:
### Link

https://moeshouse.com/collections/star-feather?srsltid=AfmBOoq0ePaepzl432UUYJg7y8ea0mTEAP7C2ZZRUoA_T858ql6Z3_TR

### Database entry

{"id":56,"type":"Router","ieeeAddr":"0xa4c138733e64f1e1","nwkAddr":64204,"manufId":4417,"manufName":"_TZE200_zo0cfekv","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"...","65506":53,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_zo0cfekv","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"ep

### Zigbee2MQTT version

 2.6.0 

### Comments

**Device support request: Moes Star Feather EU (TS0601, 3-gang)**

- **Manufacturer**: Tuya / Moes
- **Manufacturer name (fingerprint)**: `_TZE200_zo0cfekv`
- **Model ID**: `TS0601`
- **Region/variant**: EU front plate (same hardware/DPs as US SFL02-Z-3, only different fingerprint)

**Zigbee2MQTT version**
- Zigbee2MQTT: [x.y.z]
- zigbee-herdsman: [x.y.z]
- Node.js: [x.y.z]

**Database entry (from data/database.db)**
```json
{"id":56,"type":"Router","ieeeAddr":"0xa4c138733e64f1e1","nwkAddr":64204,"manufId":4417,"manufName":"_TZE200_zo0cfekv","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"...","65506":53,"65508":0,"modelId":"TS0601","manufacturerName":"_TZE200_zo0cfekv","powerSource":1,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"ep

### External definition

```shell
export default {
  zigbeeModel: ['TS0601'],
  model: 'TS0601',
  vendor: '_TZE200_zo0cfekv',
  description: 'Automatically generated definition',
  extend: [],
  meta: {},
};
```

### What does/doesn't work with the external definition?

The device pairs successfully and is visible in Zigbee2MQTT.
However, no switches are exposed (no state_l1, state_l2, etc.).
The device only shows up as "Not supported".

Expected behaviour: same as Moes SFL02-Z-3 (US version), since hardware (I think) is the same.

mattplate, 2025-09-17:
I investigated this EU variant in depth. The device is functionally identical to the already supported US Moes SFL02-Z-3 (3-gang Star Feather switch), but exposes itself with a different manufacturerName in the fingerprint.

- US version fingerprint: `_TZE200_tzyy0rtq` (and others) → already supported as SFL02-Z-3
- EU version fingerprint: `_TZE200_zo0cfekv` → currently not recognized

When adding an external converter with this fingerprint, the device works with the exact same datapoints as the US version:
- DP 24/25/26 → state_l1/state_l2/state_l3
- DP 18/19/20 → mode_l1/mode_l2/mode_l3 (switch/scene)
- DP 30/31/32 → countdown timers
- DP 36 → backlight_mode
- DP 37 → indicator_status
- DP 38 → power_on_behavior
- DP 103/104 → induction_mode / vibration_mode
- DP 1/2/3 → action scene_1/2/3

Therefore, adding the EU fingerprint `_TZE200_zo0cfekv` to the existing SFL02-Z-3 definition should be sufficient for full support out of the box.

Thanks for considering!

Koenkk, 2025-09-17:
`_TZE200_zo0cfekv` is already supported in the dev branch.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-09-17 by Koenkk