Koenkk/zigbee2mqtt #28681: Add support to Zigbee Model TS0601 - Zigbee Manufacturer _TZE284_wdfurkoa - COVER

felipesgavioli, 2025-09-20:
### Link

https://www.zigbee2mqtt.io/devices/ZM25R1.html

### Database entry

{"id":31,"type":"Router","ieeeAddr":"0x880f62fffe31b6b9","nwkAddr":26428,"manufId":4098,"manufName":"_TZE284_wdfurkoa","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184,0,60672],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE284_wdfurkoa","powerSource":1,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{},"lastSeen":1758388101640}

### Zigbee2MQTT version

2.6.1-1

### Comments

NOT Support to cover zemismart ZM25R1 

Zigbee Model: TS0601

Zigbee Manufacturer: _TZE284_wdfurkoa

### External definition

```shell
NOT Support to cover zemismart ZM25R1 

Zigbee Model: TS0601

Zigbee Manufacturer: _TZE284_wdfurkoa
```

### What does/doesn't work with the external definition?

NOT Support to cover zemismart ZM25R1 

Zigbee Model: TS0601

Zigbee Manufacturer: _TZE284_wdfurkoa

Koenkk, 2025-09-22:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

closed 2025-09-22 by Koenkk

felipesgavioli, 2025-09-22:
0x880f62fffe31b6b9
Not supported: generated

ZigBee Model
TS0601
_TZE284_wdfurkoa (Automatically generated definition)
Model
[TS0601](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)
[Unsupported](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html)
MQTT
zigbee2mqtt/0x880f62fffe31b6b9

Bjk8kds, 2025-09-24:
> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

You'll need to move to the Dev branch if you want to experience the changes now. Or you can wait until early next month.