Koenkk/zigbee2mqtt #28682: Aqara SSM-U01 not working with conbee iii latest firmware

erlenfas, 2025-09-20:
### What happened?

With the latest firmware update on conbee iii 0x26550900, aqara single switch relay with neutral does not provide measurement of voltage, current, power and energy. These properties are provided as null. Running z2m 2.6.1.

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

0x26550900

### Adapter

Conbee iii

### Setup

HAOS on rpi 4

erlenfas, 2025-09-24:
Hi, yes pretty sure about this - might also be the mqtt version influencing. As far as my research goes it seems it’s the lumi-specific properties not getting interpreted from the deconz data. The power/energy reporting relies on lumi.fromZigbee.lumi_specific.

rommess, 2025-11-23:
same config as @erlenfas (Sonoff Zigbee 3.0 dongle with Ember 8.0.2 [GA])

[image]

via the Dev Console is OK
I've two of these: same behavor