Koenkk/zigbee2mqtt #28687: Configuration is not consistent with adapter state/backup

GrumpyMeow, 2025-09-21:
### What happened?

Ah crap... This is the third time i get this error. Two days i also had this error. And one week before also. Every time i had to repair all my devices. I didn't had this issue before a week ago.
I thought the error would be related to having my Sonoff P Zigbee Dongle connected to my OpenWRT via ser2net. So two days ago i'd moved the dongle back to the machine running Zigbee2mqtt, but alas i now have the issue again.

As it is now just 15mins ago, i'm able to specify what i've done:
1. I'd added/paired a Tuya zigbee airquality box.
2. I'd changed the name of this device
3. I'd changed the "Last seen" from "disable" to "ISO_8601_local".
4. I'd scrolled down on the page after changing the "LAST_SEEN" optie and chose to "Save". (full xpath: /html/body/div/div[2]/main/div/div/div/div/div/div[2]/div/form/div[2]/button)
5. I then noticed the red-colored "Restart" button in the top-right (full xpath: /html/body/div/div[2]/nav/div/button[2]).
6. Then i noticed the mqtt-state "zigbee2mqtt/bridge/state" was offline.
7. I then navigated to the Zigbee2mqtt Homeassistant-addon which indeed was stopped.
8. I scrolled down and saw the error i saw two times before.

When looking at the error it states that the **Configured** Network key changed:
```
Network Key: configured=01030507090b0d0f00020406080a0c0d, adapter:active=75c844512cd7adbc7a36dcd7af3abe55 adapter:preconfigured=75c844512cd7adbc7a36dcd7af3abe55, adapter:alternate=75c844512cd7adbc7a36dcd7af3abe55
```

I've also

The complete error from log:
```
Starting Zigbee2MQTT without watchdog.
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: Configuration is not consistent with adapter state/backup!
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: - PAN ID: configured=31352, adapter=31352
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: - Extended PAN ID: configured=36c75a32b93d45ec, adapter=36c75a32b93d45ec
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: - Network Key: configured=01030507090b0d0f00020406080a0c0d, adapter:active=75c844512cd7adbc7a36dcd7af3abe55 adapter:preconfigured=75c844512cd7adbc7a36dcd7af3abe55, adapter:alternate=75c844512cd7adbc7a36dcd7af3abe55
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: - Channel List: configured=25, adapter=25
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: Please update configuration to prevent further issues.
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: If you wish to re-commission your network, please remove coordinator backup at /config/zigbee2mqtt/coordinator_backup.json.
[2025-09-21 12:20:32] error: 	zh:adapter:zstack:manager: Re-commissioning your network will require re-pairing of all devices!
[2025-09-21 12:20:32] error: 	z2m: Error while starting zigbee-herdsman
[2025-09-21 12:20:32] error: 	z2m: Failed to start zigbee-herdsman
[2025-09-21 12:20:32] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-09-21 12:20:32] error: 	z2m: Exiting...
[2025-09-21 12:20:32] error: 	z2m: Error: startup failed - configuration-adapter mismatch - see logs above for more information
    at ZnpAdapterManager.determineStrategy (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:248:23)
    at ZnpAdapterManager.start (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:61:26)
    at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:154:16)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
    at Zigbee.start (/app/lib/zigbee.ts:67:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
```

### What did you expect to happen?

I didnt expect the configured network to be modified.

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

20250321

### Adapter

Sonoff Dongle P

### Setup

* Home Assistant addon
* Proxmox VM Home Assistant
* Proxmox VE host

GrumpyMeow, 2025-09-21:
I'm closing this issue as i was able to resolve it myself. Hopefully others will be helped with the given solution.

closed 2025-09-21 by GrumpyMeow

andrewjfreyer, 2026-03-04:
I also experienced this error, and was able to find a solution I couldn't find referenced elsewhere. Here were the general steps I stumbled through to finally enter a state where I could restore from backup. The fundamental problem appears to be that the alternate network key and the main network key become desynchronized somehow, and herdsman expects them to be identical. For me, z2m 2.80 + SLZB-06 SMLight coordinator with via the linux installation method. 

These are my process notes, I hope they are helpful to someone hoping to avoid re-pairing devices:  

- stop z2m (systemctl, docker, whatever)
- ssh and manually back up the `data` directory. 
- delete the database.db and coordinator_backup.json
- edit configuration.yaml to include `onboarding: true`
- launch z2m & begin onboarding, select to generate new keys
- let the database get populated, and wait for z2m to fully initialize. You will see 0 devices, despite the devices listed in the configuration.
- stop the z2m process  
- edit configuration.yaml to include `onboarding: true` (again)
- edit configuration.yaml to include your old network keys, pan, and ex pan
- launch z2m & begin onboarding (again) (if you get a 'port in use' error, use `sudo ss -lptn 'sport = :8080'` and `kill -9 [port]` with the pid that returns from that command)
- the z2m process should stall about 20 - 30 seconds in, and time out saying that the network key is in use <- this is what we want
- now, physically move your coordinator's antenna or otherwise isolate the RF environment. If you can power cycle all router devices, that may work, but I was able to remove my SMLight SLZB-06's antenna. 
- Once the coordinator is RF isolated in some way, re-launch the onboarding process by spinning up z2m and loading the onboarding splash page
- **NOW** when you onboard, the coordinator will not receive interference, and will allow itself to be configured with your old network key. 
- Once z2m fully launches, you will see 0 devices, despite that you have an untouched configuration.yaml and you have successfully returned to your old network credentials. 
- Stop z2m, replace the data directory to restore from backup. 

This process is the only thing that worked for me. I did **not** have to re-pair any of my 196 devices.

GigiPompieru, 2026-03-05:
> Somehow i suspect that when onboarding and choosing a new network-key, it is not stored permenantly on the zigbee adapter or not updating the confguration.yaml file. So when changing a setting, it creates a backup and then fails because the old key is fetched from the adapter and stored in the backup.

Can this be confirmed and hopefully fixed? Why would you onboard on an existing network?