Koenkk/zigbee2mqtt #28695: [Wrong device]: ZG-227ZS Temperature & Humidity sensor

Youri3246, 2025-09-21:
### Link

https://www.zigbee2mqtt.io/devices/ZG-227ZL.html#tuya-zg-227zl

### Model

ZG-227ZS

### Description

Temperature & Humidity sensor

### Vendor

HOBEIAN

### Picture (link)

https://www.zigbee2mqtt.io/devices/ZG-227Z.html

### Database entry

{"id":61,"type":"EndDevice","ieeeAddr":"0xa4c138a82500a94b","nwkAddr":24203,"manufId":4742,"manufName":"HOBEIAN","powerSource":"Battery","modelId":"ZG-227Z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":770,"inClusterList":[0,3,1026,1029,1],"outClusterList":[3],"clusters":{"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200,"batteryVoltage":30}},"msTemperatureMeasurement":{"attributes":{"measuredValue":2580}},"msRelativeHumidity":{"attributes":{"measuredValue":4830}},"genBasic":{"attributes":{"manufacturerName":"HOBEIAN","zclVersion":3,"appVersion":147,"modelId":"ZG-227Z","powerSource":3}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":147,"stackVersion":2,"hwVersion":1,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1946894323},"lastSeen":1758487318779}

Peter376, 2025-11-13:
Hi @frech-tech ,
The default “workaround” is to just accept the model mismatch and enjoy the device. 
I paired all four devices now and they all report a 3-4% higher relative humidity than my Sonoff SNZB-02D (1 unit) and Tuya TS0201 (3 units). So I dialed in “-4” for calibration in the “specific settings”:

[image]

And for now that’s how I go with them. They are usable.

QuickJack, 2026-11-15:
I do have a similar HOBEIAN sensor which is labeled "ZG-227ZS" on the back cover. It seems to have a different temperature sensor:

[image]

Zigbee2Mqtt reports it as:
Zigbee-Modell
ZG-227Z

HOBEIAN
Definition (Zigbee2MQTT)

Firmware-ID
0123052026
23052026

Unfortunately, resolutions are artificially restricted to 0.1°C and 1%. Interestingly, this sensor sends both, manuSpecificTuya and standard ZCL clusters. The latter include full resolutions for temperature and humidity. While temperature is still rounded to 0.1%, using the msRelativeHumidity cluster allows for higher resolution humidity measurements.

To be able to use the higher resolutions I created the following converter:
```
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['ZG-227Z'],
    model: 'ZG-227Z',
    vendor: 'HOBEIAN',
    description: 'Temperature & humidity sensor (ZCL high precision)',
    fingerprint: [
        {modelID: 'ZG-227Z', manufacturerName: 'HOBEIAN'},
    ],
    extend: [
        m.temperature(),
        m.humidity(),
        m.battery(),
    ],
}; 
```