Koenkk/zigbee2mqtt #28743: Z2M refuses to start on Windows + Sonoff ZBDongle-E

schmurtzm, 2025-09-26:
### What happened?

I'm struggling to use my Sonoff Dongle-E. Z2M systemtically reboot it in bootloader mode and refuse to connect to it
It's not the first time that I use these dongles with Z2M but this this time I'm not able to make it work.
I'm running Z2M 2.6.1 on Windows.

### What did you expect to happen?

Something like that:
```
[2025-09-24 09:57:02] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-09-24 09:57:02] info:     zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-09-24 09:57:02] info:     zh:ember:uart:ash: Serial port opened
[2025-09-24 09:57:02] info:     zh:ember:uart:ash: ======== ASH starting ========
[2025-09-24 09:57:03] info:     zh:ember:uart:ash: ======== ASH connected ========
[2025-09-24 09:57:03] info:     zh:ember:uart:ash: ======== ASH started ========
[2025-09-24 09:57:03] info:     zh:ember:ezsp: ======== EZSP started ========
```

### How to reproduce it (minimal and precise)

Full logs below.

My config:

```
version: 4
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.0.111:1883
serial:
  port: COM4
  adapter: ember
  baudrate: 115200
  rtscts: false
```

Previously my Dongle-E had a CH9102 serial controler, this time my new Dongle-E has a CP2102N 🤔 

I'm able to flash the dongle online with  https://darkxst.github.io/silabs-firmware-builder/
I'm also able to flash it with xterminal.
I have tested many other firmwares than the 7.4.4.
My last test has been done with ember-zli: I made a full app clear and then flashed it with Nerivec's firmware. 
However I still have the same problem. 🙁

I can easily reproduce the problem from ember-zli  itself: 
if I run `ember-zli stack` or `ember-zli monitor`, it starts the same process as Z2M:
```
[2025-09-25 06:33:52.426] info:         zh:ember:ezsp: ======== EZSP starting ========
[2025-09-25 06:33:52.426] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-09-25 06:33:52.427] info:         zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-09-25 06:33:52.432] info:         zh:ember:uart:ash: Serial port opened
[2025-09-25 06:33:52.432] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-09-25 06:33:57.842] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-09-25 06:33:57.843] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-09-25 06:34:03.278] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-09-25 06:34:03.280] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-09-25 06:34:08.730] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-09-25 06:34:08.731] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-09-25 06:34:14.159] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-09-25 06:34:14.160] info:         zh:ember:uart:ash: ======== ASH starting ========
    Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
```

I remak that after running Z2M or `ember-zli stack` then my dongle enter in bootloader mode each time...

For example:
Before running z2m:
```
python.exe Elelabs_EzspFwUtility.py probe -p COM11
[... 308 lines cut ...]
[2025-09-25 15:07:18] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 3800/5000
[2025-09-25 15:07:18] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 3900/5000
[2025-09-25 15:07:18] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4000/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4100/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4200/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4300/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4400/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4500/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4600/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4700/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4800/5000
[2025-09-25 15:07:19] debug: 	zh:ember:uart:ash: Waiting for RSTACK... 4900/5000
[2025-09-25 15:07:20] error: 	z2m: Error while starting zigbee-herdsman
[2025-09-25 15:07:20] error: 	z2m: Failed to start zigbee-herdsman
[2025-09-25 15:07:20] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-09-25 15:07:20] error: 	z2m: Exiting...
[2025-09-25 15:07:20] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (H:\my test\z2mtest\\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@6.0.4\node_modules\zigbee-herdsman\src\adapter\ember\adapter\emberAdapter.ts:665:19)
    at EmberAdapter.start (H:\my test\z2mtest\\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@6.0.4\node_modules\zigbee-herdsman\src\adapter\ember\adapter\emberAdapter.ts:1547:24)
    at Controller.start (H:\my test\z2mtest\\zigbee2mqtt\node_modules\.pnpm\zigbee-herdsman@6.0.4\node_modules\zigbee-herdsman\src\controller\controller.ts:133:29)
    at Zigbee.start (H:\my test\z2mtest\\zigbee2mqtt\lib\zigbee.ts:67:27)
    at Controller.start (H:\my test\z2mtest\\zigbee2mqtt\lib\controller.ts:101:13)
    at start (H:\my test\z2mtest\\zigbee2mqtt\index.js:149:5)

```

schmurtzm, 2025-10-01:
[image]

Nerivec, 2025-10-01:
Can you give this custom firmware build a try? Make sure to use `rtscts=true` with it (and if testing with Z2M, set `rtscts: true` in config).
https://github.com/Nerivec/silabs-firmware-builder/actions/runs/18174241063
(have to extract, the GBL is inside)

_Note that if this firmware does not work, you will have to enter the bootloader manually (with the buttons) to reflash it._

If the firmware does not work, can you also confirm if you still get the response in vscode serial monitor if you uncheck the DTR and RTS boxes?

schmurtzm, 2025-10-01:
Still the same: 

```
ember-zli stack
[2025-10-01 10:25:53.104] info:         cli: Data folder: C:\Users\Schmurtz\ember-zli.
v Path: COM11, Baudrate: 115200, RTS/CTS: false. Use this config? Yes
[2025-10-01 10:25:53.871] info:         zh:ember:ezsp: ======== EZSP starting ========
[2025-10-01 10:25:53.871] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-10-01 10:25:53.872] info:         zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-10-01 10:25:53.876] info:         zh:ember:uart:ash: Serial port opened
[2025-10-01 10:25:53.877] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-10-01 10:25:59.332] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-10-01 10:25:59.333] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-10-01 10:26:04.754] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-10-01 10:26:04.757] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-10-01 10:26:10.194] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-10-01 10:26:10.194] info:         zh:ember:uart:ash: ======== ASH starting ========
[2025-10-01 10:26:15.629] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-10-01 10:26:15.631] info:         zh:ember:uart:ash: ======== ASH starting ========
    Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
```

Still working in serial monitor when uncheck uncheck DTR and RTS:

[image]

The output is very different in serial monitor if I dont uncheck DTR and RTS.

Nerivec, 2025-10-01:
Yes, it sounds more and more like a driver problem.
Have you tried uninstalling the CP driver completely? (I can't remember if Windows defaults to a built-in driver or not, but if it does, maybe that one works... 😰)

schmurtzm, 2025-10-01:
As ugly workaround, if I add a virtual COM port that mirrors the output of the CP2102N, then it works:

1. Install **com0com**  
2. Create a linked pair of virtual ports: `CNCA0 <-> CNCB0` → they will appear as `COM12` and `COM13`, for example  
3. Install **hub4com**  
4. Run the following command:

```bash
   hub4com --baud=115200 --octs=off \\.\COM11 \\.\COM12
```

And then ember-zli and Z2M should work:

```
ember-zli stack
[2025-10-02 12:55:55.193] info:         cli: Data folder: C:\Users\Schmurtz\ember-zli. 
v Path: COM13, Baudrate: 115200, RTS/CTS: false. Use this config? Yes 
[2025-10-02 12:55:57.856] info:         zh:ember:ezsp: ======== EZSP starting ========
[2025-10-02 12:55:57.857] info:         zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-10-02 12:55:57.857] info:         zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-10-02 12:55:57.859] info:         zh:ember:uart:ash: Serial port 
[2025-10-02 12:55:57.859] info:         zh:ember:uart:ash: ======== ASH starting 
[2025-10-02 12:55:58.993] info:         zh:ember:uart:ash: ======== ASH connected 
[2025-10-02 12:55:59.055] info:         zh:ember:uart:ash: ======== ASH started 
[2025-10-02 12:55:59.055] info:         zh:ember:ezsp: ======== EZSP started 
[2025-10-02 12:55:59.074] info:         ember: NCP EZSP protocol version (16) lower than Host. 
[2025-10-02 12:55:59.083] info:         ember: NCP version: {"ezsp":16,"revision":"8.0.3 [GA]","build":581,"major":8,"minor":0,"patch":3,"special":0,"type":170}
```

schmurtzm, 2025-11-05:
Hi, I've shared my configuration in the first post, but really nothing special in it.

About the ch340 I think you can use any of them, for my test I'm not sure but I think that I've used this old one:

https://s.click.aliexpress.com/e/_EJQnZS0
But any other usb-serial should work.

roboshs, 2025-11-05:
> Hi, I've shared my configuration in the first post, but really nothing special in it.
> 
> About the ch340 I think you can use any of them, for my test I'm not sure but I think that I've used this old one:
> 
> https://s.click.aliexpress.com/e/_EJQnZS0 But any other usb-serial should work.

zbdonglee_zigbee_ncp_8.0.3.0_sw_flow_115200.gbl just use this firmware from darkxst it will start working. i have tried it with the latest silicon labs driver v11.4.0 as well as v7.6.0, both work now. it was definitely the firmware issue with 8.0.2 that came with the sonoff dongle official flasher. just flash the 8.0.3.0 and no more problems. my z2m addon version is 2.6.3-1 which is the latest too. the sonoff official flasher does not even show the 8.0.3 version i had to take it from darkxst's github and used the flasher from his github itself.

roboshs, 2025-11-05:
> I've tested previously the 8.0.3 from darkxst ,directly from the [online flash method](https://darkxst.github.io/silabs-firmware-builder/), without success (this is the one I used to make my test with a separate USB-UART).

thats suprising. since i had the same error as you for days. same exact chip cp2102, i tried downgrading the firmware to 7.4.4, 7.5.4, and 8.0.2. then i ended up trying to downgrade z2m itself till version 2.6.0. but to no avail. as soon as i did the 8.0.3 it straight away worked. though i deleted all my previous z2m folder configurations, the addon itself, backup etc. the z2m folder itself. maybe just try it again. my setup is windows 11 with hyper v, usbipd sharing the usb port to haos. drivers the latest ones from silicon labs.

Nerivec, 2025-11-05:
Have you maybe tried with the latest pre-release firmware (8.2.2)?
https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2025.6.2-pre0

roboshs, 2025-11-06:
@Nerivec sonoff_zbdonglee_zigbee_ncp_8.2.2.0_115200_sw_flow.gbl ---- Works. 
[13:50:13]``` INFO: Preparing to start...
[13:50:13] INFO: Socat not enabled
[13:50:13] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2025-11-06 13:50:14] info: 	z2m: Logging to console, file (filename: log.log)
[2025-11-06 13:50:14] info: 	z2m: Starting Zigbee2MQTT version 2.6.3 (commit #unknown)
[2025-11-06 13:50:14] info: 	z2m: Starting zigbee-herdsman (6.3.2)
[2025-11-06 13:50:14] info: 	zh:adapter:discovery: Matched adapter: {"path":"/dev/ttyUSB0","manufacturer":"Itead","serialNumber":"c4a39b8d6153ef11baec28e0174bec31","pnpId":"usb-Itead_Sonoff_Zigbee_3.0_USB_Dongle_Plus_V2_c4a39b8d6153ef11baec28e0174bec31-if00-port0","vendorId":"10c4","productId":"ea60"} => ember: 4
[2025-11-06 13:50:14] info: 	zh:ember: Using default stack config.
[2025-11-06 13:50:14] info: 	zh:ember: ======== Ember Adapter Starting ========
[2025-11-06 13:50:14] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2025-11-06 13:50:14] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2025-11-06 13:50:14] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2025-11-06 13:50:14] info: 	zh:ember:uart:ash: Serial port opened
[2025-11-06 13:50:14] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2025-11-06 13:50:15] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2025-11-06 13:50:15] info: 	zh:ember:uart:ash: ======== ASH started ========
[2025-11-06 13:50:15] info: 	zh:ember:ezsp: ======== EZSP started ========
[2025-11-06 13:50:15] info: 	zh:ember: Adapter version info: {"ezsp":18,"revision":"8.2.2 [GA]","build":436,"major":8,"minor":2,"patch":2,"special":0,"type":170}
[2025-11-06 13:50:16] info: 	zh:ember: [STACK STATUS] Network up.
[2025-11-06 13:50:16] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2025-11-06 13:50:16] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2025-11-06 13:50:16] info: 	z2m: zigbee-herdsman started (resumed)
[2025-11-06 13:50:16] info: 	z2m: Coordinator firmware version: '{"meta":{"build":436,"ezsp":18,"major":8,"minor":2,"patch":2,"revision":"8.2.2 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2025-11-06 13:50:16] info: 	z2m: Currently 0 devices are joined.
[2025-11-06 13:50:16] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2025-11-06 13:50:16] info: 	z2m: Connected to MQTT server
[2025-11-06 13:50:16] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2025-11-06 13:50:16] info: 	z2m: Started frontend on port 8099
[2025-11-06 13:50:16] info: 	z2m: Zigbee2MQTT started!
[2025-11-06 13:50:21] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/1221051039810110150109113116116_0x8c8b48fffe45d018/connection_state/config', payload '{"default_entity_id":"binary_sensor.zigbee2mqtt_bridge_connection_state","device":{"hw_version":"EmberZNet 8.2.2 [GA]","identifiers":["zigbee2mqtt_bridge_0x8c8b48fffe45d018"],"manufacturer":"Zigbee2MQTT","model":"Bridge","name":"Zigbee2MQTT Bridge","sw_version":"2.6.3"},"device_class":"connectivity","entity_category":"diagnostic","name":"Connection state","object_id":"zigbee2mqtt_bridge_connection_state","origin":{"name":"Zigbee2MQTT","sw":"2.6.3","url":"https://www.zigbee2mqtt.io"},"payload_off":"offline","payload_on":"online","state_topic":"zigbee2mqtt/bridge/state","unique_id":"bridge_0x8c8b48fffe45d018_connection_state_zigbee2mqtt","value_template":"{{ value_json.state }}"}'

Nerivec, 2025-11-06:
This is not Windows, not a COM port. From the logs, this is HA add-on.

Nerivec, 2025-12-25:
If it's RPI-5, I doubt this is the same issue (not Windows).
Make sure nothing else can lock the port (ZHA, flasher or some other app that could be trying to open a serial port).
Do you have some `debug` logs from the startup?
What firmware are you using (source repo & version)?

devkaiwang, 2025-12-25:
It seems random. Some times it works but others not. I restarted RPI, now it works. I wonder if it is related to my local converter code on human presence sensor?

Firmware is Revision: 8.2.2 [GA], it was from https://github.com/Nerivec/silabs-firmware-builder/releases, but not sure which build.

Here is the error log when issue happens:

```log
[2025-12-25 23:36:29] info: 	z2m: Successfully interviewed '0xa4c138d39e1288fd', device has successfully been paired
[2025-12-25 23:36:29] info: 	z2m: Device '0xa4c138d39e1288fd' is supported, identified as: Tuya PIR 24GHz human presence sensor (TS0601-PIR-Sensor)
[2025-12-25 23:36:29] info: 	z2m: Configuring '0xa4c138d39e1288fd'
[2025-12-25 23:36:29] debug: 	zh:controller:endpoint: ZCL command 0xa4c138d39e1288fd/1 genBasic.read(["manufacturerName","zclVersion","appVersion","modelId","powerSource",65534], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false})
[2025-12-25 23:36:29] debug: 	zh:ember: ~~~> [ZCL to=0xa4c138d39e1288fd:57334 apsFrame={"profileId":260,"clusterId":0,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":0,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":10,"commandIdentifier":0}]
[2025-12-25 23:36:29] debug: 	zh:ember:ezsp: ===> [FRAME: ID=52:"SEND_UNICAST" Seq=52 Len=37]
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=4 frmRx=5](ackRx=4)
[2025-12-25 23:36:29] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/binary_sensor/0xa4c138d39e1288fd/presence/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"default_entity_id":"binary_sensor.0xa4c138d39e1288fd_presence","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c138d39e1288fd"],"manufacturer":"Tuya","model":"PIR 24GHz human presence sensor","model_id":"TS0601-PIR-Sensor","name":"0xa4c138d39e1288fd","via_device":"zigbee2mqtt_bridge_0x08b95ffffec632e2"},"device_class":"occupancy","object_id":"0xa4c138d39e1288fd_presence","origin":{"name":"Zigbee2MQTT","sw":"2.7.1","url":"https://www.zigbee2mqtt.io"},"payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/0xa4c138d39e1288fd","unique_id":"0xa4c138d39e1288fd_presence_zigbee2mqtt","value_template":"{{ value_json.presence }}"}'
[2025-12-25 23:36:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"definition":{"description":"PIR 24GHz human presence sensor","exposes":[{"access":1,"description":"Indicates whether the device detected presence","label":"Presence","name":"presence","property":"presence","type":"binary","value_off":false,"value_on":true},{"access":1,"category":"diagnostic","description":"Link quality (signal strength)","label":"Linkquality","name":"linkquality","property":"linkquality","type":"numeric","unit":"lqi","value_max":255,"value_min":0}],"model":"TS0601-PIR-Sensor","options":[],"source":"external","supports_ota":false,"vendor":"Tuya"},"friendly_name":"0xa4c138d39e1288fd","ieee_address":"0xa4c138d39e1288fd","status":"successful","supported":true},"type":"device_interview"}'
[2025-12-25 23:36:29] info: 	z2m:mqtt: MQTT publish: topic 'homeassistant/sensor/0xa4c138d39e1288fd/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state","value_template":"{{ value_json.state }}"}],"default_entity_id":"sensor.0xa4c138d39e1288fd_linkquality","device":{"hw_version":1,"identifiers":["zigbee2mqtt_0xa4c138d39e1288fd"],"manufacturer":"Tuya","model":"PIR 24GHz human presence sensor","model_id":"TS0601-PIR-Sensor","name":"0xa4c138d39e1288fd","via_device":"zigbee2mqtt_bridge_0x08b95ffffec632e2"},"enabled_by_default":false,"entity_category":"diagnostic","icon":"mdi:signal","name":"Linkquality","object_id":"0xa4c138d39e1288fd_linkquality","origin":{"name":"Zigbee2MQTT","sw":"2.7.1","url":"https://www.zigbee2mqtt.io"},"state_class":"measurement","state_topic":"zigbee2mqtt/0xa4c138d39e1288fd","unique_id":"0xa4c138d39e1288fd_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4](ackRx=4 frmTx=5)
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=4 frmNum=5](frmRx=5) Added to rxQueue
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=6](ackRx=4)
[2025-12-25 23:36:29] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=155:"ZIGBEE_KEY_ESTABLISHMENT_HANDLER" Seq=51 Len=14]
[2025-12-25 23:36:29] debug: 	zh:ember:ezsp: ezspZigbeeKeyEstablishmentHandler: partner=0xa4c138d39e1288fd status=TC_RESPONDED_TO_KEY_REQUEST
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA]
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5](ackRx=4 frmTx=5)
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=5 frmNum=6](frmRx=6) Added to rxQueue
[2025-12-25 23:36:29] debug: 	zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=5)
[2025-12-25 23:36:29] debug: 	zh:ember:ezsp: <=== [FRAME: ID=52:"SEND_UNICAST" Seq=52 Len=10]
[2025-12-25 23:36:29] debug: 	zh:ember:ezsp: ~~~> [SENT type=DIRECT apsSequence=160 messageTag=11 status=OK]
[... cut ...]
    at EmberAdapter.initEzsp (/app/node_modules/.pnpm/zigbee-herdsman@7.0.4/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:666:19)
    at EmberAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@7.0.4/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1539:24)
    at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@7.0.4/node_modules/zigbee-herdsman/src/controller/controller.ts:143:29)
    at Zigbee.start (/app/lib/zigbee.ts:70:27)
    at Controller.start (/app/lib/controller.ts:101:13)
    at start (/app/index.js:149:5)
```

Ric121, 2026-01-08:
Did you guys made any progress on that issue? I have the same behaviour, z2m not connecting on windows with a Sonoff Dongle-E. Curiously it worked fine on a HA Green (Linux).
On windows I tried

- z2m last release 2.7.2, z2m 2.6.1 (to use the same zigbee-herdsman as you), openHAB 5.1.0. All failed in the connection to the device (see traces below)
- Firmware versions: 7.4.0, 7.5.0, 8.02, official flashed with Sonoff on-line flashing tool, 8.2.2 sonoff_zbdonglee_zigbee_ncp_8.2.2.0_115200_sw_flow.gbl found on nerivec flashed with sil-labs tool,
- Drivers versions: CP210x_Universal_Windows_Driver, CP210x_VCP_Windows, from https://www.silabs.com/software-and-tools/usb-to-uart-bridge-vcp-drivers?tab=downloads
- ember, ezsp, all kind of settings combinations in the configuration file and in the driver (although I suppose the settings from the driver in windows UI don't have importance).

I always get the following which is ore or less the same as you:
Starting Zigbee2MQTT without watchdog.
[2026-01-08 19:13:35] debug:    z2m: Removing old log directory 'C:\Zigbee2MQTT\zigbee2mqtt\data\log\2026-01-08.00-40-10'
[2026-01-08 19:13:35] info:     z2m: Logging to console, file (filename: log.log)
[2026-01-08 19:13:35] debug:    z2m: Loaded state from file C:\Zigbee2MQTT\zigbee2mqtt\data\state.json
[2026-01-08 19:13:35] info:     z2m: Starting Zigbee2MQTT version 2.7.2 (commit #3a49c957)
[2026-01-08 19:13:35] info:     z2m: Starting zigbee-herdsman (8.0.1)
[2026-01-08 19:13:35] debug:    z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":27615,\"extendedPanID\":[106,6,247,6,60,254,185,177],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"C:\\\\Zigbee2MQTT\\\\zigbee2mqtt\\\\data\\\\database.db\",\"databaseBackupPath\":\"C:\\\\Zigbee2MQTT\\\\zigbee2mqtt\\\\data\\\\database.db.backup\",\"backupPath\":\"C:\\\\Zigbee2MQTT\\\\zigbee2mqtt\\\\data\\\\coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":true,\"path\":\"COM9\",\"adapter\":\"ember\"},\"adapter\":{\"disableLED\":false}}"'
[2026-01-08 19:13:35] debug:    zh:adapter:discovery: Connected devices: [{"path":"COM3","manufacturer":"Microsoft","pnpId":"BTHENUM\\{00001101-0000-1000-8000-00805F9B34FB}_LOCALMFG&0000\\7&3528DEA4&0&000000000000_00000002","friendlyName":"Standard Serial over Bluetooth link (COM3)"},{"path":"COM4","manufacturer":"Microsoft","pnpId":"BTHENUM\\{00001101-0000-1000-8000-00805F9B34FB}_VID&0001000A_PID&FFFF\\7&3528DEA4&0&C0288D849973_C00000000","friendlyName":"Standard Serial over Bluetooth link (COM4)"},{"path":"COM9","manufacturer":"Silicon Laboratories","serialNumber":"FEF5A4BA7F12F011B6C20A14773D9DA9","pnpId":"USB\\VID_10C4&PID_EA60\\FEF5A4BA7F12F011B6C20A14773D9DA9","locationId":"Port_#0002.Hub_#0002","friendlyName":"Silicon Labs CP210x USB to UART Bridge (COM9)","vendorId":"10C4","productId":"EA60"}]
[2026-01-08 19:13:35] debug:    zh:adapter:discovery: Unable to match USB adapter: ember | COM9
[2026-01-08 19:13:35] info:     zh:ember: Using default stack config.
[2026-01-08 19:13:35] debug:    zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":27615,"extendedPanID":[106,6,247,6,60,254,185,177],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"COM9","adapter":"ember"},"adapter":{"disableLED":false},"databasePath":"C:\\Zigbee2MQTT\\zigbee2mqtt\\data\\database.db","databaseBackupPath":"C:\\Zigbee2MQTT\\zigbee2mqtt\\data\\database.db.backup","backupPath":"C:\\Zigbee2MQTT\\zigbee2mqtt\\data\\coordinator_backup.json"}'
[2026-01-08 19:13:35] info:     zh:ember: ======== Ember Adapter Starting ========
[2026-01-08 19:13:35] info:     zh:ember:ezsp: ======== EZSP starting ========
[2026-01-08 19:13:35] info:     zh:ember:uart:ash: ======== ASH Adapter reset ========
[2026-01-08 19:13:35] debug:    zh:ember:uart:ash: Opening serial port with {"path":"COM9","baudRate":115200,"rtscts":true,"autoOpen":false,"parity":"none","stopBits":1,"xon":false,"xoff":false}
[2026-01-08 19:13:35] info:     zh:ember:uart:ash: Serial port opened
[2026-01-08 19:13:35] info:     zh:ember:uart:ash: ======== ASH starting ========
[2026-01-08 19:13:35] debug:    zh:ember:uart:ash: ---> [FRAME type=RST]
[2026-01-08 19:13:35] debug:    zh:ember:uart:ash: Waiting for RSTACK... 0/5000
[2026-01-08 19:13:35] debug:    zh:ember:uart:ash: Waiting for RSTACK... 100/5000
[2026-01-08 19:13:35] debug:    zh:ember:uart:ash: Waiting for RSTACK... 200/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 300/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 400/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 500/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 600/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 700/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 800/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 900/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1000/5000
[2026-01-08 19:13:36] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1100/5000
[2026-01-08 19:13:37] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1200/5000
[2026-01-08 19:13:37] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1300/5000
[2026-01-08 19:13:37] debug:    zh:ember:uart:ash: Waiting for RSTACK... 1400/5000
[... cut ...]
[2026-01-08 19:13:40] debug:    zh:ember:uart:ash: Waiting for RSTACK... 4000/5000
[2026-01-08 19:13:40] debug:    zh:ember:uart:ash: Waiting for RSTACK... 4100/5000
[2026-01-08 19:13:40] debug:    zh:ember:uart:ash: Waiting for RSTACK... 4200/5000
[... 37 lines cut ...]
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  pan_id: 6754
  ext_pan_id:
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
    - 221
frontend:
  enabled: false
  port: 8080
homeassistant:
  enabled: false
onboarding: false