Koenkk/zigbee2mqtt #28747: [Wrong device]: 2 Button Scene Switch

Mrsash2020, 2025-09-27:
### Link

https://www.aliexpress.com/item/1005006083341888.html

### Model

None

### Description

Zigbee Switch Push Button Switch 2Gang 6 Scene

### Vendor

NA

### Picture (link)

https://ae-pic-a1.aliexpress-media.com/kf/A8fa22cec06fc43bea39eb916f8e6a4bdR.jpg_.webp

### Database entry

https://www.zigbee2mqtt.io/devices/33951_33948.html#awox-33951%252F33948

### Notes

Ok So my new 2 button scene switch gets identified as [AwoX](https://www.zigbee2mqtt.io/supported-devices/#v=AwoX)

[image]

Koenkk, 2025-11-02:
For the non-working ones, can you stop z2m, replace all occurrences of `TLSR82xx` with `_TZ3210_3ulg9kpo`, start z2m and see if it works?

MattiaCarli, 2025-11-07:
Hello, i have the same problem different Aliexpress LINK: https://it.aliexpress.com/item/1005005814154185.html?spm=a2g0o.order_list.order_list_main.11.30a118028Ledn1&gatewayAdapt=glo2ita

[image]

import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['TLSR82xx'],
    model: 'TLSR82xx',
    vendor: 'TELINK',
    description: 'Automatically generated definition',
    extend: [m.deviceEndpoints({"endpoints":{"1":1,"2":2}}), m.battery(), m.iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]})],
};

{
    "description": "LED white",
    "exposes": [
        {
            "features": [
                {
                    "access": 7,
                    "description": "On/off state of this light",
                    "label": "State",
                    "name": "state",
                    "property": "state",
                    "type": "binary",
                    "value_off": "OFF",
                    "value_on": "ON",
                    "value_toggle": "TOGGLE"
                },
                {
                    "access": 7,
                    "description": "Brightness of this light",
                    "label": "Brightness",
                    "name": "brightness",
                    "property": "brightness",
                    "type": "numeric",
                    "value_max": 254,
                    "value_min": 0
                }
            ],
            "type": "light"
        },
        {
            "access": 2,
            "description": "Triggers an effect on the light (e.g. make light blink for a few seconds)",
            "label": "Effect",
            "name": "effect",
            "property": "effect",
            "type": "enum",
            "values": [
                "blink",
                "breathe",
                "okay",
                "channel_change",
                "finish_effect",
                "stop_effect"
            ]
        },
        {
            "access": 7,
            "category": "config",
            "description": "Controls the behavior when the device is powered on after power loss",
            "label": "Power-on behavior",
            "name": "power_on_behavior",
            "property": "power_on_behavior",
            "type": "enum",
            "values": [
                "off",
                "on",
                "toggle",
                "previous"
            ]
        },
        {
            "access": 1,
            "category": "diagnostic",
            "description": "Link quality (signal strength)",
            "label": "Linkquality",
            "name": "linkquality",
            "property": "linkquality",
            "type": "numeric",
            "unit": "lqi",
            "value_max": 255,
            "value_min": 0
        }
    ],
    "model": "33951/33948",
    "options": [
        {
            "access": 2,
            "description": "Controls the transition time (in seconds) of on/off, brightness, color temperature (if applicable) and color (if applicable) changes. Defaults to `0` (no transition).",
            "label": "Transition",
            "name": "transition",
            "property": "transition",
            "type": "numeric",
            "value_min": 0
        },
        {
            "access": 2,
            "description": "State actions will also be published as 'action' when true (default false).",
            "label": "State action",
            "name": "state_action",
            "property": "state_action",
            "type": "binary",
            "value_off": false,
            "value_on": true
        }
    ],
    "source": "native",
    "supports_ota": false,
    "vendor": "AwoX"
}

ntzb, 2026-02-27:
this is what I came up with - `/config/zigbee2mqtt/external_converters/scene_switch.mjs`:
```
import * as m from "zigbee-herdsman-converters/lib/modernExtend";

export default {
  fingerprint: [{ manufacturerName: "TELINK", modelID: "TLSR82xx" }],
  model: "Scene_Switch_2gang",
  vendor: "Tuya",
  description: "2-button scene switch",
  extend: [
    m.deviceEndpoints({ endpoints: { 1: 1, 2: 2 } }),
    m.battery(),
    m.onOff({ powerOnBehavior: false, endpointNames: ["1", "2"] }),
  ],
  meta: { multiEndpoint: true },
};
```
it allows only 2 scenes, not 6 - single press ***only***, double/long press does not work.