Koenkk/zigbee2mqtt #28748: [Wrong device]: Tuya 19DZT

TobiasDax, 2025-09-27:
### Link

#

### Model

19DZT

### Description

Door or Window Sensor

### Vendor

Tuya

### Picture (link)

https://www.zigbee2mqtt.io/images/devices/19DZT.png

### Database entry

Unknown

### Notes

When I pair the door sensor it shows up in z2m as this device: https://www.zigbee2mqtt.io/devices/ZC-W1.html

I re paired the device several times but that’s not doing anything.

alexcfer2, 2025-09-28:
Same issue here when using Tuya door sensor from image below. It used to work well but stopped since updating Zigbee2MQTT. Now it's recognised as a temp sensor.

[image]

eeveelution1313, 2025-10-01:
Same problem, shows up in 2.6 as contact sensor [TS0203](https://www.zigbee2mqtt.io/devices/TS0203.html#tuya-ts0203) but in 2.6.1 it shows up as [ZC-W1](https://www.zigbee2mqtt.io/devices/ZC-W1.html)