Koenkk/zigbee2mqtt #28751: SNZB-02D loose connnection when routed via Ikea bulb

lattemacchiato, 2025-09-27:
### What happened?

Hi, 
Sorry if this issue might already have been discussed, but I couldn't find any solution.
Problem: I run several SNZB-02D via z2m. Some of them loose connnection randomly (found several reports in this issue...). By some intensive testing, I could reproduce the connnection problems and nail it down: SNZB-02D loose connnection when they are routed via an Ikea bulb (in my case TRADFRI bulb E27 WW clear 250lm, Modell LED1842G3, latest firmware) and z2m is restarted. After restart, z2m sends a mqtt message for those SNZB-02D exactly once, then never again. When I remove the bulb, the SNZB-02D find a new router device (plug, another bulb,...) and z2m starts sending mqtt again. Alternative, when I do a re-pairing of the SNZB-02D, routing via the bulb starts to work again. So, problem seems to be the routing functionality between bulb and z2m after restart. Since it seems not to be possible to deactivate the routing functionality on the bulb, my only solution so far is removing the bulb from z2m :-(
Might be a bug in the bulb's firmware, or can it be fixed in z2m? 
Thanks, 
Paul

### What did you expect to happen?

I expected, that routing via bulb works also after restart. 

### How to reproduce it (minimal and precise)

-  place SNZB-02D near bulb to force routing via bulb
- restart z2m
- Monitor mqtt: only one initial mqtt msg from SNZB-02D received, but no subsequent ones

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

7.4.4

### Adapter

Sonoff ZBDongle-E, ember driver

### Setup

Docker Version on Intel nuc, Ubuntu
Linux fhem-nuc 6.8.0-83-generic #83-Ubuntu SMP PREEMPT_DYNAMIC Fri Sep  5 21:46:54 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
Ubuntu 24.04.3 LTS

### Device `database.db` entry

{"id":6,"type":"Router","ieeeAddr":"0xccccccfffed46289","nwkAddr":65061,"manufId":4476,"manufName":"IKEA of Sweden","powerSource":"Mains (single phase)","modelId":"TRADFRI bulb E27 WW clear 250lm","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":257,"inClusterList":[0,3,4,5,6,8,4096,64636],"outClusterList":[5,25,32,4096],"clusters":{"genBasic":{"attributes":{"modelId":"TRADFRI bulb E27 WW clear 250lm","manufacturerName":"IKEA of Sweden","powerSource":1,"zclVersion":3,"appVersion":32,"stackVersion":99,"hwVersion":2,"dateCode":"20181203","swBuildId":"2.3.093"}},"genOnOff":{"attributes":{"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":99,"hwVersion":2,"dateCode":"20181203","swBuildId":"2.3.093","zclVersion":3,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":1324213189},"lastSeen":1756837691637}

chris-1243, 2025-09-29:
I have several SNZB-02P/D/WD in my network and they are mostly connected to some IKEA bulbs (LED2109G6) and I have no issues at all.

They do survive a restart of Z2M and work reliably.

Have you checked the link quality between your SNZB-02D and your router ?

lattemacchiato, 2025-09-29:
Hi,

I also tried with another Ikea bulb (LED1732G11). Reproducible: If a bulb is present at reboot (!), z2m stops sending mqtt messages from SNZB-02D located near the bulb. Disconnecting the bulb (i.e. remove from power completely) brings the SNZB-02D back to live, at the latest after reboot. Without bulb present, the SNZB-02D "survive" every reboot. 
I can connect the bulb after reboot - everything continues to run fine, z2m continues to send SNZB-02D mqtt messages - till reboot.
With bulb connected, some don't. LQ is about 160 (should even be better with bulb connected, since bulb is located near the SNZB-02Ds).
No other devices show issues (plugs, DoorSensors, TempSensors, Buttons...).
Since two Ikea bulbs show the same "behaviour" it seems not to be related to a specific bulb...

chris-1243, 2025-09-29:
It might be an incompatibility between those two brands which might happen. I would not bet Ikea and Sonoff strictly follow the Zigbee spec.

If you have an other light bulb from an other brand to try instead of Ikea, this also could solve your issue and narrow the problem. If it is firmware related, there is nothing Z2M might do.

You might also have interference at this specific emplacement. By experience, I know SNZB-02D/P/WD work well with Hue bulbs.

OptiMael, 2025-10-03:
I had the same problem with my SNZB-02P and 02D.
Removing the Ikea bulbs from the Zigbee network seems to be the only solution at the moment.

chris-1243, 2025-10-03:
@OptiMael 

Which model of Ikea bulbs ?

closed 2025-12-10 by github-actions[bot]