Koenkk/zigbee2mqtt #28767: Connection Problems with Zigbee2MQTT and SLZB-MR3

NoPro200, 2025-09-30:
### What happened?

After 1-5 days, the stick loses its connection to Zigbee2MQTT and devices also go offline randomly. After that, I have to completely restart the stick (power cycle) so that a new connection is possible.

### What did you expect to happen?

that this does not happen

### How to reproduce it (minimal and precise)

it just happens

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

v3.0.6

### Adapter

SLZB-MR3

### Setup

zigbee2mqtt is installed as an add-on in Home Assistant (Home Assistant runs as a VM on Proxmox).

### Debug log

[2025-09-30 05:15:05] debug: 	z2m: Loaded state from file /config/zigbee2mqtt/state.json
[2025-09-30 05:15:05] info: 	z2m: Starting Zigbee2MQTT version 2.6.1 (commit #unknown)
[2025-09-30 05:15:05] info: 	z2m: Starting zigbee-herdsman (6.0.4)
[2025-09-30 05:15:05] debug: 	z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":6754,\"extendedPanID\":[221,221,221,221,221,221,221,221],\"channelList\":[20],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/config/zigbee2mqtt/database.db\",\"databaseBackupPath\":\"/config/zigbee2mqtt/database.db.backup\",\"backupPath\":\"/config/zigbee2mqtt/coordinator_backup.json\",\"serialPort\":{\"baudRate\":115200,\"rtscts\":true,\"path\":\"tcp://192.168.30.14:7638\",\"adapter\":\"zstack\"},\"adapter\":{\"disableLED\":false,\"transmitPower\":20}}"'
[2025-09-30 05:15:05] debug: 	zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[20]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"tcp://192.168.30.14:7638","adapter":"zstack"},"adapter":{"disableLED":false,"transmitPower":20},"databasePath":"/config/zigbee2mqtt/database.db","databaseBackupPath":"/config/zigbee2mqtt/database.db.backup","backupPath":"/config/zigbee2mqtt/coordinator_backup.json"}'
[2025-09-30 05:15:05] info: 	zh:zstack:znp: Opening TCP socket with 192.168.30.14:7638
[2025-09-30 05:17:21] error: 	zh:zstack:znp: Socket error Error: connect ETIMEDOUT 192.168.30.14:7638
[2025-09-30 05:17:21] error: 	z2m: Error while starting zigbee-herdsman
[2025-09-30 05:17:21] error: 	z2m: Failed to start zigbee-herdsman
[2025-09-30 05:17:21] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-09-30 05:17:21] error: 	z2m: Exiting...
[2025-09-30 05:17:21] error: 	z2m: Error: Error while opening socket
    at Socket.<anonymous> (/app/node_modules/.pnpm/zigbee-herdsman@6.0.4/node_modules/zigbee-herdsman/src/adapter/z-stack/znp/znp.ts:160:24)
    at Socket.emit (node:events:530:35)
    at emitErrorNT (node:internal/streams/destroy:170:8)
    at emitErrorCloseNT (node:internal/streams/destroy:129:3)
    at processTicksAndRejections (node:internal/process/task_queues:90:21)

Sometimes there is also an error with a 6000ms timeout.

Popoboxxo, 2025-09-30:
Got the same Problem with my SLZB-MR1. 

"
Starting Zigbee2MQTT without watchdog.
[2025-07-28 21:11:35] error: z2m: Error while starting zigbee-herdsman
[2025-07-28 21:11:35] error: z2m: Failed to start zigbee-herdsman
[2025-07-28 21:11:35] error: z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start_crashes-runtime.html for possible solutions
[2025-07-28 21:11:35] error: z2m: Exiting...
[2025-07-28 21:11:35] error: z2m: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
at ZStackAdapter.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:114:27)
at Controller.start (/app/node_modules/.pnpm/zigbee-herdsman@4.3.1/node_modules/zigbee-herdsman/src/controller/controller.ts:133:29)
at Zigbee.start (/app/lib/zigbee.ts:68:27)
at Controller.start (/app/lib/controller.ts:101:13)
at start (/app/index.js:149:5)"

I have already talked with the SMLIGHT Support.

"
1. Shut down Zigbee2MQTT/ZHA and power off your previous Zigbee coordinator.
2. Power off all your Zigbee routers for at least 15 seconds. You can do this at once by turning off the electricity in the whole building.
3. Power on back all your Zigbee routers.
4. Power on your new SLZB-06/06p7 or 06p10 adapter.
5. Start your Zigbee2MQTT or ZHA.
6. It may take some time (e.g., 5-10 minutes) for your Zigbee devices to come back online again with the new Zigbee coordinator.
"

But this isn't really helpful..

NoPro200, 2025-09-30:
I accidentally closed it.

closed 2025-09-30 by NoPro200