Koenkk/zigbee2mqtt #28768: [New device support]:

arevatech, 2025-09-30:
### Link

https://www.zigbee2mqtt.io/devices/PN6.html

### Database entry

i cant found it..

### Zigbee2MQTT version

2.6.1-1

### Comments

i added this device, but is show like unsupported. 

[image]

### External definition

```shell
it show me like unsupported device,
```

### What does/doesn't work with the external definition?

it show me like unsupported device,

Koenkk, 2025-09-30:
You are using ZHA which is a different project than Zigbee2MQTT, please create an issue in the ZHA issue tracker.

closed 2025-09-30 by Koenkk

arevatech, 2025-10-02:
Hey, no. i use Zigbee2MQTT project in HA.

[image]