Koenkk/zigbee2mqtt #28777: Aqara T1 Cube not working with ZBDongle-E 8.02

erlenfas, 2025-09-30:
### What happened?

Pairing seems not successful and the link button does not give response after pairing.

### What did you expect to happen?

Link button to give response, state to be updated.

### How to reproduce it (minimal and precise)

Pair Aqara T1 Pro cube to ZBDongle-E with zigbee2mqtt.

### Zigbee2MQTT version

2.6.1

### Adapter firmware version

8.0.2 [GA]

### Adapter

ZBDongle-E

### Setup

Home Assistant on RPI5 HAOS

Nerivec, 2025-10-01:
Can you provide a `debug` log?
https://www.zigbee2mqtt.io/guide/configuration/logging.html#debugging

Nerivec, 2025-10-02:
We need `debug` level. This is `info`, it does not contain enough information for this kind of bug.